osrf/mbzirc #73: ignition library incompatible?

femust, 2022-03-29:
Hey guys,

during the compilation of the simulation I encountered the following error:

[image]

I noticed that this is related to the API of the library "gazebo" version that you use. Can you give some hints on which one we should use? Since even If I installed ignition Fortress I can't see for example WorldAngularAcceleration, have a look here:

[image]

while here, in the API the function WorldAngularAcceleration is [present](https://ignitionrobotics.org/api/gazebo/6.7/classignition_1_1gazebo_1_1Link.html) 

Cheers,

mjcarroll, 2022-03-29:
Can you check which version of `libignition-gazebo6` you have installed?

`apt-cache showpkg libignition-gazebo6`

and 

`cat /usr/include/ignition/gazebo6/ignition/gazebo/config.hh`

iche033, 2022-03-29:
That looks like an older version of ignition-gazebo. You can try installing the latest version and see if it fixes the issue:

```
sudo apt update
sudo apt install libignition-gazebo6-dev
```

femust, 2022-03-30:
@mjcarroll 
[image]

```
sudo apt update
sudo apt install libignition-gazebo6-dev
```
but this worked, I was updating only the ign Fortress version without realizing that libraries need to be updated as well 🤦🏻  thanks! 

One thing since I have run a simple example
```
ros2 launch ros_ign_gazebo ign_gazebo.launch.py ign_args:="-v 4 -r simple_demo.sdf"
```
and the sea can't be rendered and it seems that physics also does not work since the vessel is at the bottom of the sea. Should other libraries be updated as well? If yes which one? (same thing for the coast environment)

[image]

mjcarroll, 2022-03-30:
It's very likely that you are also missing updates to some of the dependencies.

The easiest thing to do would be to `sudo apt update && sudo apt upgrade` to update your entire system.

If there are things you dont want updated, then:

```
sudo apt update &&
sudo apt install \
libignition-cmake2-dev \
libignition-common4-dev \
libignition-fuel-tools7-dev \
libignition-gazebo6-dev \
libignition-gui6-dev \
libignition-sensors6-dev \
libignition-physics5-dev \
libsdformat12-dev
```

Should cover the libraries that you need.

femust, 2022-03-30:
After updating still the same while sea. Logs from simple_demo:

```
andrzej@femust:~/code_workspace/MBZIRC$ ros2 launch ros_ign_gazebo ign_gazebo.launch.py ign_args:="-v 4 -r simple_demo.sdf"
[INFO] [launch]: All log files can be found below /home/andrzej/.ros/log/2022-03-30-19-13-59-870534-femust-5244
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ign gazebo-1]: process started with pid [5246]
[ign gazebo-1] [Wrn] [Filesystem.cc:140] The path [/tmp/ign/mbzirc/logs] does not refer to a directory nor to a file
[ign gazebo-1] [Msg] Ignition Gazebo Server v6.7.0
[ign gazebo-1] [Msg] Loading SDF world file[/home/andrzej/code_workspace/MBZIRC/install/mbzirc_ign/share/mbzirc_ign/worlds/simple_demo.sdf].
[ign gazebo-1] [Dbg] [Physics.cc:801] Loaded [ignition::physics::dartsim::Plugin] from library [/usr/lib/x86_64-linux-gnu/ign-physics-5/engine-plugins/libignition-physics-dartsim-plugin.so]
[ign gazebo-1] [Dbg] [SystemManager.cc:49] Loaded system [ignition::gazebo::systems::Physics] for entity [1]
[ign gazebo-1] [Msg] Create service on [/world/simple_demo/create]
[ign gazebo-1] [Msg] Remove service on [/world/simple_demo/remove]
[ign gazebo-1] [Msg] Pose service on [/world/simple_demo/set_pose]
[ign gazebo-1] [Msg] Light configuration service on [/world/simple_demo/light_config]
[ign gazebo-1] [Msg] Physics service on [/world/simple_demo/set_physics]
[ign gazebo-1] [Msg] SphericalCoordinates service on [/world/simple_demo/set_spherical_coordinates]
[ign gazebo-1] [Msg] Enable collision service on [/world/simple_demo/enable_collision]
[ign gazebo-1] [Msg] Disable collision service on [/world/simple_demo/disable_collision]
[ign gazebo-1] [Msg] Material service on [/world/simple_demo/visual_config]
[ign gazebo-1] [Dbg] [SystemManager.cc:49] Loaded system [ignition::gazebo::systems::UserCommands] for entity [1]
[ign gazebo-1] [Dbg] [Sensors.cc:406] Configuring Sensors system
[ign gazebo-1] [Dbg] [Sensors.cc:325] SensorsPrivate::Run
[ign gazebo-1] [Dbg] [SystemManager.cc:49] Loaded system [ignition::gazebo::systems::Sensors] for entity [1]
[ign gazebo-1] [Dbg] [Sensors.cc:305] SensorsPrivate::RenderThread started
[ign gazebo-1] [Dbg] [Sensors.cc:195] Waiting for init
[ign gazebo-1] [Dbg] [SystemManager.cc:49] Loaded system [ignition::gazebo::systems::AirPressure] for entity [1]
[ign gazebo-1] [Dbg] [SystemManager.cc:49] Loaded system [ignition::gazebo::systems::Altimeter] for entity [1]
[ign gazebo-1] [Dbg] [SystemManager.cc:49] Loaded system [ignition::gazebo::systems::Imu] for entity [1]
[ign gazebo-1] [Dbg] [SystemManager.cc:49] Loaded system [ignition::gazebo::systems::Magnetometer] for entity [1]
[ign gazebo-1] [Msg] Serving particle emitter information on [/world/simple_demo/particle_emitters]
[ign gazebo-1] [Dbg] [SystemManager.cc:49] Loaded system [ignition::gazebo::systems::ParticleEmitter2] for entity [1]
[ign gazebo-1] [Dbg] [SystemManager.cc:49] Loaded system [ignition::gazebo::systems::SceneBroadcaster] for entity [1]
[ign gazebo-1] [Dbg] [SystemManager.cc:49] Loaded system [ignition::gazebo::systems::WindEffects] for entity [1]
[ign gazebo-1] [Msg] MBZIRC log path: /tmp/ign/mbzirc/logs
[ign gazebo-1] [Msg] Run duration set to 3600 seconds.
[ign gazebo-1] [GUI] [Wrn] [Application.cc:753] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:102:33: QML Settings: Failed to initialize QSettings instance. Status code is: 1
[ign gazebo-1] [GUI] [Wrn] [Application.cc:753] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:102:33: QML Settings: The following application identifiers have not been set: QVector("organizationName", "organizationDomain")
[ign gazebo-1] [Msg] Geofence boundary min: -70 -70 -25, max: 70 70 121.92
[ign gazebo-1] [Msg] Setup duration set to 600 seconds.
[ign gazebo-1] [Msg] Target vessel: Vessel A
[ign gazebo-1] [Msg]   Small object: small_box_a
[ign gazebo-1] [Msg]   Large object: large_box_a
[ign gazebo-1] [Msg] Starting MBZIRC
[ign gazebo-1] [Dbg] [SystemManager.cc:49] Loaded system [mbzirc::GameLogicPlugin] for entity [1]
[ign gazebo-1] [Msg] Using Pierson-Moskowitz spectrum sampling wavefield model 
[ign gazebo-1] [Msg] Using Pierson-Moskowitz spectrum sampling wavefield model 
[ign gazebo-1] [Msg] Input Parameters:
[ign gazebo-1] [Msg] model:     PMS
[ign gazebo-1] [Msg] number:     3
[ign gazebo-1] [Msg] scale:      1.1
[ign gazebo-1] [Msg] angle:      0.4
[ign gazebo-1] [Msg] steepness:  0
[ign gazebo-1] [Msg] amplitude:  0
[ign gazebo-1] [Msg] period:     5
[ign gazebo-1] [Msg] direction:  1 0
[ign gazebo-1] [Msg] tau:  2
[ign gazebo-1] [Msg] gain:  0.3
[ign gazebo-1] [Msg] Derived Parameters:
[ign gazebo-1] [Msg] amplitudes:  0.0363502, 0.0392277, 0.0385703, 
[ign gazebo-1] [Msg] wavenumbers: 0.133171, 0.161136, 0.194975, 
[ign gazebo-1] [Msg] omegas:      1.1424, 1.25664, 1.3823, 
[ign gazebo-1] [Msg] periods:     [Msg] 5.5, [Msg] 5, [Msg] 4.54545, [Msg] 
[ign gazebo-1] [Msg] phases:      0, 0, 0, 
[ign gazebo-1] [Msg] steepnesses: 0, 0, 0, 
[ign gazebo-1] [Msg] directions:  [Msg] 0.921061 -0.389418; [Msg] 1 0; [Msg] 0.921061 0.389418; [Msg] 
[ign gazebo-1] [Dbg] [SystemManager.cc:49] Loaded system [ignition::gazebo::systems::WaveVisual] for entity [27]
[ign gazebo-1] [Msg] Using Pierson-Moskowitz spectrum sampling wavefield model 
[ign gazebo-1] [Msg] Using Pierson-Moskowitz spectrum sampling wavefield model 
[... 276 lines cut ...]
[... cut ...] te<std::mutex> > > > >, std::__detail::_Select1st, std::equal_to<std::vector<unsigned long, std::allocator<unsigned long> > >, ignition::gazebo::v6::detail::ComponentTypeHasher, std::__detail::_Mod_range_hashing, std::__detail::_Default_ranged_hash, std::__detail::_Prime_rehash_policy, std::__detail::_Hashtable_traits<true, false, true> >::clear()
[ign gazebo-1] Segmentation fault (Address not mapped to object [0x7f2ad99fb048])
[INFO] [ign gazebo-1]: process has finished cleanly [pid 64302]

```

iche033, 2022-03-30:
The white sea textrue could be outdated ign-rendering library or graphics driver issue.  

A couple of things to try:
* check if you have `libignition-rendering6-dev` version 6.2.1? If not, you can apt install it
* test the shaders example world in ign-gazebo by running`ign gazebo -v 4 shader_param.sdf`. You should see waves there.

The ODE internal error is likely the result of the target vessel falling to the bottom. That could be that the plugins are not working. What branch are you on in `mbzirc`?

You can try clearing the model cache for that particular model: 

```
rm -r ~/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/vessel\ a/
```

femust, 2022-03-30:
1) ```check if you have libignition-rendering6-dev version 6.2.1? If not, you can apt install it``` 
 [image]
2)  ```ign gazebo -v 4 shader_param.sdf``` seems to be working  
[image]
3) ```rm -r ~/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/vessel\ a/```
still the same. 
[image]

branch ```main``` everywhere

iche033, 2022-03-30:
looks like the libraries are up-to-date and our shaders work fine with your graphics card. I'm not sure why you're getting the same result. Last thing I can think of is to do a clean rebuild of you colcon workspace

Alternatively, you can try running the sim with [docker](https://docs.docker.com/engine/install/ubuntu/) and see if that works for you: https://github.com/osrf/mbzirc#docker-setup

If so, you an inspect the environment in docker to see what's different between that and your setup.

femust, 2022-03-30:
Ok, so it seems in the docker is working (I will go wht's going step by step with docker), but still there are some problems with simulator in the docker since after:
```
ros2 launch mbzirc_ign spawn.launch.py name:=quadrotor_1 world:=simple_demo model:=mbzirc_quadrotor type:=uav x:=1 y:=2 z:=1.05 R:=0 P:=0 Y:=0 slot0:=mbzirc_hd_camera slot1:=mbzirc_rgbd_camera```

![image](https://user-images.githubusercontent.com/26916586/160930475-7a57ee7a-e9ad-45a4-8590-67ff0ebbf9cf.png)
+seg fault

iche033, 2022-03-30:
it looks like you pull the latest ignition debs in the docker container. It's caused by a release made yesterday. We're fixing it. In the mean time we just merged a [workaround](https://github.com/osrf/mbzirc/pull/74) in this repo. You can `git pull` now in the `mbzirc` source code of your docker container to get the latest changes, and `colcon build --merge-install` for it to take effect.

femust, 2022-03-30:
ok, one question, do you have some schedule for the final release, or are you going to update it continuously?

FYI: i made it run locally by following docker, but frankly speaking I'm not sure what has changed since the initial phase, thanks!

iche033, 2022-03-31:
The next release will be on April 15th. In the mean time, we'll continue to make changes here in `main` branch so it'll be unstable. After April 15th, we'll make changes based on feedback and bug reports but I expect them to be smaller patches.

iche033, 2022-03-31:
and also good to hear that's it's now working locally for you!

closed 2022-03-31 by iche033