osrf/mbzirc #98: Docker container communications

zorawar12, 2022-04-26:
Hi, I was trying to run separate containers for each robot. However, I am not sure how can we make it all visible in a single display environment. Shall we create another container for that, as you showed during the webinar? Is it possible to get some more insights into the docker container communication pipeline? Where do we need to give the IP of individual containers from the docker network...and those details.

iche033, 2022-04-26:
From the high level, there will be a simulator container, multiple bridge containers, and multiple robot containers. I've temporarily uploaded the diagram to this page: https://github.com/osrf/mbzirc/wiki/Infrastructure. We will add more information over the next few weeks as we work on the infrastructure setup for evaluation.

> However, I am not sure how can we make it all visible in a single display environment.

Not sure the issue you're running into. For local testing, we will look into providing Dockerfiles for building a bridge container and an example team solution (robot) container.

zorawar12, 2022-04-26:
> From the high level, there will be a simulator container, multiple bridge containers, and multiple robot containers. I've temporarily uploaded the diagram to this page: [...]

In this image, let's assume that I create a docker network and run two separate images in that network wherein one image I run USV and in another a quad, so if I just want to send a message from USV image to quad image using the method explained [here](https://github.com/osrf/mbzirc/wiki/Communications), how would the quad docker image find the USV image just through the model name as an address, is there some backend code that gets IP of the images for communication?

> Not sure the issue you're running into. For local testing, we will look into providing Dockerfiles for building a bridge image and an example team solution (robot) image.

This will be great! Looking forward to seeing this.

iche033, 2022-04-27:
> how would the quad docker image find the USV image just through the model name as an address, is there some backend code that gets IP of the images for communication?

yeah so when a robot is spawned, it registers itself as an endpoint using with its name. When the USV sends a message to a quad, the message goes to the simulator first, who will run it through the comms model and determine if it should be dropped or not. If the message can be delivered, the simulator sends that message to the quad.

zorawar12, 2022-05-06:
>> There is no direct communication between the robot containers.

The simulator will be running in a different container than the robot itself, so how would the robot container know where is the simulator container? Is it possible for you to add an example of that? I don't see that there is any robot-bridge docker container available in the package to create communication between the simulator and the robot container.

zorawar12, 2022-05-07:
I got it to work. I think I was missing out on the presence of the new DDS mechanism in the backend in ROS2.

closed 2022-05-07 by zorawar12