osrf/mbzirc #99: Fixed wing

zorawar12, 2022-04-26:
The fixed-wing freezes when with RGB-D camera payload at slot-1 after reaching a height of 146m whereas it keeps flying without any payload. I tried changing flightTime, assuming that the camera is consuming the battery, but that didn't seem to work.

iche033, 2022-04-26:
146m exceeds the max flight height allowed (400 ft / 121.92m) so the simulator freezes the UAV.  There is some buffer allowed when the vehicle moves outside of the boundary but if it keeps going, the 'kill switch' is activated which freezes the vehicle

In our latest `main` branch (to be released in v4), we also made it so that the fixed wing will freeze at that height without any payload.

zorawar12, 2022-04-26:
But without a payload camera, it does moves to 500m height

zorawar12, 2022-04-26:
Okay, Thank you for the clarification...however, in our white paper, we have mentioned that we need to fly that high...

closed 2022-04-26 by iche033