osrf/mbzirc #100: Problem with launching simple demo

mandicLuka, 2022-04-27:
OS: Windows 11 + WSL2
Kernel: Linux version 5.10.60.1-microsoft-standard-WSL2 
OpenGL version: 4.2

After successful build, I tried to launch the ros simple demo using 

`ros2 launch mbzirc_ros competition_local.launch.py ign_args:="-v 4 -r simple_demo.sdf"`

Models are loaded successfully, but the following error crashes the gazebo simulator:

[ign gazebo-1] Error [parser_urdf.cc:3227] Unable to call parseURDF on robot model
[ign gazebo-1] Error [parser.cc:848] parse as old deprecated model file failed.
[ign gazebo-1] [GUI] [Err] [Model.hh:73] Unable to unserialize sdf::Model

iche033, 2022-04-27:
The crash could be due to something else. The error messages you saw there are misleading and described in the [Troubleshooting](https://github.com/osrf/mbzirc/wiki/Troubleshooting) page. We are working on suppressing those messages.

One thing to try is to launching the simulator without any MBZIRC components and see if it still crashes, e.g.

```
ign gazebo -v 4 sensors_demo.sdf
```

mandicLuka, 2022-04-28:
@iche033 You are right, the problem persists. Turns out this is the error that causes the crash:

terminate called after throwing an instance of 'Ogre::UnimplementedException'
  what():  OGRE EXCEPTION(9:UnimplementedException):  in GL3PlusTextureGpu::copyTo at /var/lib/jenkins/workspace/ogre-2.2-debbuilder/repo/RenderSystems/GL3Plus/src/OgreGL3PlusTextureGpu.cpp (line 677)

I tried downloading and building the new Ogre but nothing helped. Can you tell me the correct Ogre version for this to work?

iche033, 2022-04-28:
we are using a fork of ogre-next which is a few commits behind [ogre 2.2.6](https://github.com/OGRECave/ogre-next/tree/v2.2.6). You can try with 2.2.6. The error message looks like a GL driver issue and maybe related to running under WSL2.