osrf/mbzirc #101: Decoupling ROS packages from docker

fzoric8, 2022-04-28:
Hi, 

I've wanted to ask is it possible to decouple docker git repository from ROS packages.
 
The main reason to do that would be easier pulling simulation updates (`mbzirc_ros` and `mbzirc_`ign`) than constantly 
rebuilding docker images as well as docker 

Currently, `mbzirc_ros` and `mbzirc_ign` are part of `mbzirc` repository which is used to build docker image. 

Decoupling them in separate repositories would enable us to have up-to-date git repository of those packages which could then
be update inside already existing docker container.

iche033, 2022-04-28:
hi, I would like to see if we can workaround the issue without separating the repositories since some of the infra work we're doing for evaluation requires them to be in a single repo. 

I just want to check your use case here. You would like to work inside the docker container and keep mbzirc packages up-to-date without rebuilding a new docker image? You can do this without copying `mbzirc_ign` and `mbzirc_ros` into the workspace but just need to update some permissions. Not sure if the following helps with your use case:

1. Go to the `mbzirc_ws` directory, remove build directories, and change permission to `developer`

    ```
    cd mbzirc_ws
    sudo rm -fr build install log
    sudo chown -R developer:developer *    
    ```
1. Go to the `mbzirc` directory (there is a typo here but being fixed in a [PR](https://github.com/osrf/mbzirc/pull/102)) and update url to https so you can pull the latest changes

    ```
    cd ~/mbzirc_ws/src/mbzric # typo here
    git remote set-url origin https://github.com/osrf/mbzirc.git
    git pull
    ```

1. You should then be able to rebuild the workspace with latest changes

    ```
    cd ~/mbzirc_ws/
    IGNITION_VERSION=fortress colcon build --merge-install
    ```

fzoric8, 2022-04-28:
Works. 

Thanks a lot! :+1:

fzoric8, 2022-05-26:
**Regarding this issue,** 

At the time It worked, however when I built new docker image and did following: 

1. Changed permissions as stated 
2. Cloned newest `mbzirc` as instructed 
3. Tried to run `colcon build` in the `<workspace_path>` however, I get this output: 

```
CMake Error at CMakeLists.txt:11 (find_package):
  By not providing "Findmbzirc_ros.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "mbzirc_ros", but CMake did not find one.

  Could not find a package configuration file provided by "mbzirc_ros" with
  any of the following names:

    mbzirc_rosConfig.cmake
    mbzirc_ros-config.cmake

```

Then I went and checked command in docker and ran following commands: 
```
rm -rf install build 
bash -c 'source /opt/ros/galactic/setup.bash   && colcon info mbzirc_ros   && colcon build --merge-install
```

Build fails with following msg: 
```
CMake Error at CMakeLists.txt:11 (find_package):
  By not providing "Findmbzirc_ros.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "mbzirc_ros", but CMake did not find one.

  Could not find a package configuration file provided by "mbzirc_ros" with
  any of the following names:

    mbzirc_rosConfig.cmake
    mbzirc_ros-config.cmake

  Add the installation prefix of "mbzirc_ros" to CMAKE_PREFIX_PATH or set
  "mbzirc_ros_DIR" to a directory containing one of the above files.  If
  "mbzirc_ros" provides a separate development package or SDK, be sure it has
  been installed.
```

Therefore I'm not to build mbzirc ROS packages. Bear in mind that ROS is sourced and I successfuly built workspace during docker build few weeks ago. 

Help would be appreciated.

fzoric8, 2022-05-26:
Managed to build it as follows: 
```
cd <workspace>/src
git clone https://github.com/ignitionrobotics/ros_ign.git -b galactic
cd ..
colcon build
```