osrf/mbzirc #117: USV Thrusters

Glimrel, 2022-05-20:
Hello,

I have a two questions about USV.

The figure "USV specification: Thrusting Steering" shows that the orientation of the thrusters can be changed along the hull axis, while in the simulation the rotation range is across the axis. Which solution should be adopted?

Can the propellers rotate in both directions (reverse propulsion)?

Best Regards

iche033, 2022-05-23:
hmm thanks for bringing this up. I've updated the simulation to match the spec on the website in https://github.com/osrf/mbzirc/pull/119

iche033, 2022-07-19:
We are reverting the change made in https://github.com/osrf/mbzirc/pull/119 as we just got confirmation that the image on the website is incorrect.

Changes reverted in https://github.com/osrf/mbzirc/pull/179