osrf/mbzirc #118: Simulator dying issue

zorawar12, 2022-05-20:
Hello,

I am trying to get the multi-robotic systems up using docker-compose.yaml, where the docker-compose is first getting built through the build.bash and is being run through a command inside the run.bash script. However, When I am running the simulator using the compose file just to run the coastal environment, it starts and then quickly dies showing the following message:

> process has finished cleanly [pid 75]

Any guesses on why it might be dying like this?

boushakiM, 2022-05-23:
Hi, I am facing same issue!

zorawar12, 2022-05-23:
> Hi, I am facing same issue!

I haven't found any solution as of now, if I find it, I will post it here! did you find any solution yet @boushakiM ?

iche033, 2022-05-23:
hmm it sounds like gazebo is still running since it needs to be killed through ctrl+C? Are you able to attach to the docker container and check if the process is running? If any of you can share a minimal example, I can take a look.

boushakiM, 2022-05-24:
[image]
please see this

boushakiM, 2022-05-24:
[image]
and this

zorawar12, 2022-05-24:
> hmm it sounds like gazebo is still running since it needs to be killed through ctrl+C? Are you able to attach to the docker container and check if the process is running? If any of you can share a minimal example, I can take a look.

So, what we are trying to do is, Add the entry point to the Dockerfile that is provided to us. In the entry-point bash script, we add the command to launch the simulator with the simple demo or coast environment. And now we are trying to run this docker from the run.bash script where you are running the docker directly, we replace that with a compose file to run the simulation docker file.

On this, we want to add other robots as we go on developing our proof of concept.

boushakiM, 2022-05-24:
> 

Not yet! if I find something I will update you.

iche033, 2022-05-24:
@boushakiM let's continue the discussion of your issue in #121 because it seems to be a different issue.

iche033, 2022-05-24:
@m00nlab , you have the same opengl issue, please see  https://github.com/osrf/mbzirc/issues/121#issuecomment-1136117449 and we can continue the discussion in #121

iche033, 2022-05-24:
@zorawar12, I made some changes in https://github.com/osrf/mbzirc/compare/main...docker_entrypoint, which shows adding the entrypoint script to the Dockerfile. You should be able to build and run it by doing:

```
bash docker/build.bash mbzirc_sim
cd docker
bash run.bash mbzirc_sim coast.sdf
```

That should launch the `coast.sdf` environment.

I have these changes in my [fork](https://github.com/iche033/mbzirc/tree/docker_test) along with a couple of other Dockerfiles ready for a pull request once we are done updating the spawn scripts. We'll then look into providing a docker compose yaml file afterwards for testing locally

zorawar12, 2022-05-24:
> @zorawar12, I made some changes in [main...docker_entrypoint](https://github.com/osrf/mbzirc/compare/main...docker_entrypoint), which shows adding the entrypoint script to the Dockerfile. You should be able to build and run it by doing:
> 
> ```
> bash docker/build.bash mbzirc_sim
> cd docker
> bash run.bash mbzirc_sim coast.sdf
> ```
> 
> That should launch the `coast.sdf` environment.
> 
> I have these changes in my [fork](https://github.com/iche033/mbzirc/tree/docker_test) along with a couple of other Dockerfiles ready for a pull request once we are done updating the spawn scripts. We'll then look into providing a docker compose yaml file afterwards for testing locally

Thank you for replying, I have tried this already, and I am too getting the same error as @boushakiM

iche033, 2022-05-25:
the OpenGL issues in #121 have been resolved. See if any of the solutions in that ticket work for you.

zorawar12, 2022-05-27:
Yeah, the same problem I had!

Thanks!

closed 2022-05-27 by zorawar12