osrf/mbzirc #121: Trouble running simulator with error msgs related to OpenGL 3.3

iche033, 2022-05-24:
Issue ported over from #118, cc @boushakiM

[image]

[image]

iche033, 2022-05-24:
To run the simulator, a graphics card / driver with OpenGL 3.3 support is needed. It looks like you may also have problem running the base simulator Ignition. Can you share your `~/.ignition/rendering/ogre2.log`? It will contain information on the graphics card and driver that you are using so we can take a look at the setup you have.

m00nlab, 2022-05-24:
> To run the simulator, a graphics card / driver with OpenGL 3.3 support is needed. It looks like you may also have problem running the base simulator Ignition. Can you share your [...]

Hello sir,

Here is the logs from my system:

>21:38:22: Creating resource group General
21:38:22: Creating resource group Internal
21:38:22: Creating resource group Autodetect
21:38:22: SceneManagerFactory for type 'DefaultSceneManager' registered.
21:38:22: Registering ResourceManager for type Material
21:38:22: Registering ResourceManager for type Mesh
21:38:22: Registering ResourceManager for type Mesh2
21:38:22: Registering ResourceManager for type OldSkeleton
21:38:22: MovableObjectFactory for type 'ParticleSystem' registered.
21:38:22: ArchiveFactory for archive type FileSystem registered.
21:38:22: ArchiveFactory for archive type Zip registered.
21:38:22: ArchiveFactory for archive type EmbeddedZip registered.
21:38:22: DDS codec registering
21:38:22: FreeImage version: 3.18.0
21:38:22: This program uses FreeImage, a free, open source image library supporting all common bitmap formats. See http://freeimage.sourceforge.net for details
21:38:22: Supported formats: bmp,ico,jpg,jif,jpeg,jpe,jng,koa,iff,lbm,mng,pbm,pbm,pcd,pcx,pgm,pgm,png,ppm,ppm,ras,tga,targa,tif,tiff,wap,wbmp,wbm,psd,psb,cut,xbm,xpm,gif,hdr,g3,sgi,rgb,rgba,bw,exr,j2k,j2c,jp2,pfm,pct,pict,pic,3fr,arw,bay,bmq,cap,cine,cr2,crw,cs1,dc2,dcr,drf,dsc,dng,erf,fff,ia,iiq,k25,kc2,kdc,mdc,mef,mos,mrw,nef,nrw,orf,pef,ptx,pxn,qtk,raf,raw,rdc,rw2,rwl,rwz,sr2,srf,srw,sti,x3f,webp,jxr,wdp,hdp
21:38:22: OITD codec registering
21:38:22: Registering ResourceManager for type HighLevelGpuProgram
21:38:22: MovableObjectFactory for type 'Decal' registered.
21:38:22: MovableObjectFactory for type 'InternalCubemapProbe' registered.
21:38:22: MovableObjectFactory for type 'Entity' registered.
21:38:22: MovableObjectFactory for type 'Item' registered.
21:38:22: MovableObjectFactory for type 'Light' registered.
21:38:22: MovableObjectFactory for type 'Rectangle2Dv2' registered.
21:38:22: MovableObjectFactory for type 'BillboardSet' registered.
21:38:22: MovableObjectFactory for type 'ManualObject2' registered.
21:38:22: MovableObjectFactory for type 'BillboardChain' registered.
21:38:22: MovableObjectFactory for type 'RibbonTrail' registered.
21:38:22: MovableObjectFactory for type 'WireAabb' registered.
21:38:22: *-*-* OGRE Initialising
21:38:22: *-*-* Version 2.2.6 (Cerberus)
21:38:22: OverlayElementFactory for type Panel registered.
21:38:22: OverlayElementFactory for type BorderPanel registered.
21:38:22: OverlayElementFactory for type TextArea registered.
21:38:22: Registering ResourceManager for type Font
21:38:22: Loading library /usr/lib/x86_64-linux-gnu/OGRE-2.2/OGRE/RenderSystem_GL3Plus.so
21:38:22: Installing plugin: GL 3+ RenderSystem
21:38:22: OpenGL 3+ Rendering Subsystem created.
21:38:22: Found Num EGL Devices: 2
21:38:22: EGL Device: EGL_EXT_device_drm #0 /dev/dri/card0
21:38:22: Trying to init device: EGL_EXT_device_drm #0 /dev/dri/card0...
21:38:22: OGRE EXCEPTION(3:RenderingAPIException): eglInitialize failed for device EGL_EXT_device_drm #0 /dev/dri/card0 in EGLSupport::getGLDisplay at /var/lib/jenkins/workspace/ogre-2.2-debbuilder/repo/RenderSystems/GL3Plus/src/windowing/EGL/PBuffer/OgreEglPBufferSupport.cpp (line 320)
21:38:22: OGRE EXCEPTION(3:RenderingAPIException): eglInitialize failed for device EGL_EXT_device_drm #0 /dev/dri/card0 in EGLSupport::getGLDisplay at /var/lib/jenkins/workspace/ogre-2.2-debbuilder/repo/RenderSystems/GL3Plus/src/windowing/EGL/PBuffer/OgreEglPBufferSupport.cpp (line 320)
21:38:22: Destroying device: EGL_EXT_device_drm #0 /dev/dri/card0...
21:38:22: EGL Device: EGL_MESA_device_software #1
21:38:22: Trying to init device: EGL_MESA_device_software #1...
21:38:22: Created GL 4.5 context for device EGL_MESA_device_software #1
21:38:22: Destroying device: EGL_MESA_device_software #1...
21:38:22: Plugin successfully installed
21:38:22: Loading library /usr/lib/x86_64-linux-gnu/OGRE-2.2/OGRE/Plugin_ParticleFX.so
21:38:22: Installing plugin: ParticleFX
21:38:22: Particle Emitter Type 'Point' registered
21:38:22: Particle Emitter Type 'Box' registered
21:38:22: Particle Emitter Type 'Ellipsoid' registered
21:38:22: Particle Emitter Type 'Cylinder' registered
21:38:22: Particle Emitter Type 'Ring' registered
21:38:22: Particle Emitter Type 'HollowEllipsoid' registered
21:38:22: Particle Affector Type 'LinearForce' registered
21:38:22: Particle Affector Type 'ColourFader' registered
21:38:22: Particle Affector Type 'ColourFader2' registered
21:38:22: Particle Affector Type 'ColourImage' registered
21:38:22: Particle Affector Type 'ColourInterpolator' registered
21:38:22: Particle Affector Type 'Scaler' registered
21:38:22: Particle Affector Type 'Rotator' registered
[... 2830 lines cut ...]
21:38:22: GL_NV_texgen_reflection
21:38:22: GL_NV_texture_barrier
21:38:22: GL_NV_texture_env_combine4
21:38:22: GL_NV_texture_rectangle
21:38:22: GL_OES_EGL_image
21:38:22: GL_OES_read_format
21:38:22: GL_S3_s3tc
21:38:22: GL_SGIS_generate_mipmap
21:38:22: GL_SGIS_texture_border_clamp
21:38:22: GL_SGIS_texture_edge_clamp
21:38:22: GL_SGIS_texture_lod
21:38:22: GL_SUN_multi_draw_arrays
21:38:22: OGRE EXCEPTION(3:RenderingAPIException): OpenGL 3.3 is not supported. Please update your graphics card drivers. in GL3PlusRenderSystem::initialiseContext at /var/lib/jenkins/workspace/ogre-2.2-debbuilder/repo/RenderSystems/GL3Plus/src/OgreGL3PlusRender [... cut ...]
21:38:22: Added resource location '/usr/share/ignition/ignition-rendering6/ogre2/media/materials/scripts' of type 'FileSystem' to resource group 'General'
21:38:22: Added resource location '/usr/share/ignition/ignition-rendering6/ogre2/media/materials/textures' of type 'FileSystem' to resource group 'General'
21:38:22: Parsing scripts for resource group Autodetect
21:38:22: Finished parsing scripts for resource group Autodetect
21:38:22: Creating resources for group Autodetect
21:38:22: All done
21:38:22: Parsing scripts for resource group General
21:38:22: Parsing script point_cloud_point.material

m00nlab, 2022-05-24:
> > 21:38:22: GL Version = 3.1.0.0
> > 21:38:22: GL_VERSION = 3.1 Mesa 21.2.6
> > 21:38:22: GL_VENDOR = Mesa/X.org
> > 21:38:22: GL_RENDERER = llvmpipe (LLVM 12.0.0, 256 bits)
> 
> @m00nlab looks like your setup is doing software rendering. Does your machine have a discrete graphics card? I

Yes sir, all our graphics cards are discrete ones! :P

iche033, 2022-05-24:
ok it looks like it's not doing hardware accelerated rendering. If you do:

```
glxinfo  | grep render
```

it should show `direct rendering: Yes`. If it's a `No` that means your system is not the GPU for rendering. If you have nvidia card, I would recommend install nvidia drivers.

iche033, 2022-05-24:
ok can you trying install nvidia drivers? And you are running using docker, you'll need to setup `nvidia-docker2`

m00nlab, 2022-05-24:
> ok can you try installing nvidia drivers? And if you are running using docker, you'll need to setup `nvidia-docker2`

Thank you, [this](https://docs.docker.com/compose/gpu-support/) resolved my problem, I wasn't specifying GPU device correctly as mentioned here.

boushakiM, 2022-05-25:
> ok can you try installing nvidia drivers? And if you are running using docker, you'll need to setup `nvidia-docker2`

For my case, I am not using docker, the nvidia driver is already installed (I got this: nvidia-driver-510 is already the newest version (510.73.05.Obuntu0.20.04.1).
Please how can I solve this problem? I want to use the source code, not the docker.
Thank you!

boushakiM, 2022-05-25:
This is the log:

17:38:20: Creating resource group General
17:38:20: Creating resource group Internal
17:38:20: Creating resource group Autodetect
17:38:20: SceneManagerFactory for type 'DefaultSceneManager' registered.
17:38:20: Registering ResourceManager for type Material
17:38:20: Registering ResourceManager for type Mesh
17:38:20: Registering ResourceManager for type Mesh2
17:38:20: Registering ResourceManager for type OldSkeleton
17:38:20: MovableObjectFactory for type 'ParticleSystem' registered.
17:38:20: ArchiveFactory for archive type FileSystem registered.
17:38:20: ArchiveFactory for archive type Zip registered.
17:38:20: ArchiveFactory for archive type EmbeddedZip registered.
17:38:20: DDS codec registering
17:38:20: FreeImage version: 3.18.0
17:38:20: This program uses FreeImage, a free, open source image library supporting all common bitmap formats. See http://freeimage.sourceforge.net for details
17:38:20: Supported formats: bmp,ico,jpg,jif,jpeg,jpe,jng,koa,iff,lbm,mng,pbm,pbm,pcd,pcx,pgm,pgm,png,ppm,ppm,ras,tga,targa,tif,tiff,wap,wbmp,wbm,psd,psb,cut,xbm,xpm,gif,hdr,g3,sgi,rgb,rgba,bw,exr,j2k,j2c,jp2,pfm,pct,pict,pic,3fr,arw,bay,bmq,cap,cine,cr2,crw,cs1,dc2,dcr,drf,dsc,dng,erf,fff,ia,iiq,k25,kc2,kdc,mdc,mef,mos,mrw,nef,nrw,orf,pef,ptx,pxn,qtk,raf,raw,rdc,rw2,rwl,rwz,sr2,srf,srw,sti,x3f,webp,jxr,wdp,hdp
17:38:20: OITD codec registering
17:38:20: Registering ResourceManager for type HighLevelGpuProgram
17:38:20: MovableObjectFactory for type 'Decal' registered.
17:38:20: MovableObjectFactory for type 'InternalCubemapProbe' registered.
17:38:20: MovableObjectFactory for type 'Entity' registered.
17:38:20: MovableObjectFactory for type 'Item' registered.
17:38:20: MovableObjectFactory for type 'Light' registered.
17:38:20: MovableObjectFactory for type 'Rectangle2Dv2' registered.
17:38:20: MovableObjectFactory for type 'BillboardSet' registered.
17:38:20: MovableObjectFactory for type 'ManualObject2' registered.
17:38:20: MovableObjectFactory for type 'BillboardChain' registered.
17:38:20: MovableObjectFactory for type 'RibbonTrail' registered.
17:38:20: MovableObjectFactory for type 'WireAabb' registered.
17:38:20: *-*-* OGRE Initialising
17:38:20: *-*-* Version 2.2.6 (Cerberus)
17:38:20: OverlayElementFactory for type Panel registered.
17:38:20: OverlayElementFactory for type BorderPanel registered.
17:38:20: OverlayElementFactory for type TextArea registered.
17:38:20: Registering ResourceManager for type Font
17:38:20: Loading library /usr/lib/x86_64-linux-gnu/OGRE-2.2/OGRE/RenderSystem_GL3Plus.so
17:38:20: Installing plugin: GL 3+ RenderSystem
17:38:20: OpenGL 3+ Rendering Subsystem created.
17:38:20: Found Num EGL Devices: 0
17:38:20: OGRE EXCEPTION(3:RenderingAPIException): No EGL devices found! Update your GPU drivers and try again in EglPBufferSupport::EglPBufferSupport at /var/lib/jenkins/workspace/ogre-2.2-debbuilder/repo/RenderSystems/GL3Plus/src/windowing/EGL/PBuffer/OgreEglPBufferSupport.cpp (line 84)
17:38:20: EGL Headless raised an exception. Won't be available. Are drivers too old?
17:38:20: OGRE EXCEPTION(3:RenderingAPIException): No EGL devices found! Update your GPU drivers and try again in EglPBufferSupport::EglPBufferSupport at /var/lib/jenkins/workspace/ogre-2.2-debbuilder/repo/RenderSystems/GL3Plus/src/windowing/EGL/PBuffer/OgreEglPBufferSupport.cpp (line 84)
17:38:20: Plugin successfully installed
17:38:20: Loading library /usr/lib/x86_64-linux-gnu/OGRE-2.2/OGRE/Plugin_ParticleFX.so
17:38:20: Installing plugin: ParticleFX
17:38:20: Particle Emitter Type 'Point' registered
17:38:20: Particle Emitter Type 'Box' registered
17:38:20: Particle Emitter Type 'Ellipsoid' registered
17:38:20: Particle Emitter Type 'Cylinder' registered
17:38:20: Particle Emitter Type 'Ring' registered
17:38:20: Particle Emitter Type 'HollowEllipsoid' registered
17:38:20: Particle Affector Type 'LinearForce' registered
17:38:20: Particle Affector Type 'ColourFader' registered
17:38:20: Particle Affector Type 'ColourFader2' registered
17:38:20: Particle Affector Type 'ColourImage' registered
17:38:20: Particle Affector Type 'ColourInterpolator' registered
17:38:20: Particle Affector Type 'Scaler' registered
17:38:20: Particle Affector Type 'Rotator' registered
17:38:20: Particle Affector Type 'DirectionRandomiser' registered
17:38:20: Particle Affector Type 'DeflectorPlane' registered
17:38:20: Plugin successfully installed
17:38:20: CPU Identifier & Features
17:38:20: -------------------------
17:38:20:  *   CPU ID: GenuineIntel: Intel(R) Core(TM) i7-7700HQ CPU @ 2.80GHz
17:38:20:  *   Logical cores: 8
17:38:20:  *      SSE: yes
17:38:20:  *     SSE2: yes
17:38:20:  *     SSE3: yes
[... 2820 lines cut ...]
17:38:20: GL_NV_texgen_reflection
17:38:20: GL_NV_texture_barrier
17:38:20: GL_NV_texture_env_combine4
17:38:20: GL_NV_texture_rectangle
17:38:20: GL_OES_EGL_image
17:38:20: GL_OES_read_format
17:38:20: GL_S3_s3tc
17:38:20: GL_SGIS_generate_mipmap
17:38:20: GL_SGIS_texture_border_clamp
17:38:20: GL_SGIS_texture_edge_clamp
17:38:20: GL_SGIS_texture_lod
17:38:20: GL_SUN_multi_draw_arrays
17:38:20: OGRE EXCEPTION(3:RenderingAPIException): OpenGL 3.3 is not supported. Please update your graphics card drivers. in GL3PlusRenderSystem::initialiseContext at /var/lib/jenkins/workspace/ogre-2.2-debbuilder/repo/RenderSystems/GL3Plus/src/OgreGL3PlusRenderSystem.cpp (line 3364)
[... cut ...]
17:38:20: Added resource location '/usr/share/ignition/ignition-rendering6/ogre2/media/materials/scripts' of type 'FileSystem' to resource group 'General'
17:38:20: Added resource location '/usr/share/ignition/ignition-rendering6/ogre2/media/materials/textures' of type 'FileSystem' to resource group 'General'
17:38:20: Parsing scripts for resource group Autodetect
17:38:20: Finished parsing scripts for resource group Autodetect
17:38:20: Creating resources for group Autodetect
17:38:20: All done
17:38:20: Parsing scripts for resource group General
17:38:20: Parsing script point_cloud_point.material

boushakiM, 2022-05-25:
Once I used

**ros2 launch mbzirc_ros competition_local.launch.py ign_args:="-v 4 -r coast.sdf --render-engine ogre"**

I got this in the terminal: 

ros2 launch mbzirc_ros competition_local.launch.py ign_args:="-v 4 -r coast.sdf --render-engine ogre"
[INFO] [launch]: All log files can be found below /home/mbzirc/.ros/log/2022-05-25-17-40-33-990527-mbzirc-192589
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ign gazebo-1]: process started with pid [192591]
[INFO] [parameter_bridge-2]: process started with pid [192593]
[parameter_bridge-2] [INFO] [1653486034.202158373] [ros_ign_bridge]: Creating IGN->ROS Bridge: [/mbzirc/score] (ignition.msgs.Float -> std_msgs/msg/Float32) (Lazy 0): 
[parameter_bridge-2] [INFO] [1653486034.202382533] [ros_ign_bridge]: Creating IGN->ROS Bridge: [/clock] (ignition.msgs.Clock -> rosgraph_msgs/msg/Clock) (Lazy 0): 
[parameter_bridge-2] [INFO] [1653486034.202455350] [ros_ign_bridge]: Creating IGN->ROS Bridge: [/mbzirc/run_clock] (ignition.msgs.Clock -> rosgraph_msgs/msg/Clock) (Lazy 0): 
[parameter_bridge-2] [INFO] [1653486034.202474583] [ros_ign_bridge]: Creating IGN->ROS Bridge: [/mbzirc/phase] (ignition.msgs.StringMsg -> std_msgs/msg/String) (Lazy 0): 
[parameter_bridge-2] [INFO] [1653486034.202534430] [ros_ign_bridge]: Creating IGN->ROS Bridge: [/mbzirc/target/stream/status] (ignition.msgs.StringMsg -> std_msgs/msg/String) (Lazy 0): 
[ign gazebo-1] [GUI] [Wrn] [Application.cc:753] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:102:33: QML Settings: Failed to initialize QSettings instance. Status code is: 1
[ign gazebo-1] [GUI] [Wrn] [Application.cc:753] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:102:33: QML Settings: The following application identifiers have not been set: QVector("organizationName", "organizationDomain")
[ign gazebo-1] [GUI] [Wrn] [Application.cc:753] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:102:33: QML Settings: Failed to initialize QSettings instance. Status code is: 1
[ign gazebo-1] [GUI] [Wrn] [Application.cc:753] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:102:33: QML Settings: The following application identifiers have not been set: QVector("organizationName", "organizationDomain")
[ign gazebo-1] [GUI] [Wrn] [Application.cc:753] [QT] file::/Gazebo/GazeboDrawer.qml:147:3: QML Dialog: Binding loop detected for property "implicitHeight"
[ign gazebo-1] [GUI] [Wrn] [Application.cc:753] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:102:33: QML Settings: Failed to initialize QSettings instance. Status code is: 1
[ign gazebo-1] [GUI] [Wrn] [Application.cc:753] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:102:33: QML Settings: The following application identifiers have not been set: QVector("organizationName", "organizationDomain")
[ign gazebo-1] [GUI] [Wrn] [Application.cc:753] [QT] file::/Gazebo/GazeboDrawer.qml:147:3: QML Dialog: Binding loop detected for property "implicitHeight"
[ign gazebo-1] [GUI] [Wrn] [Application.cc:753] [QT] file::/EntityContextMenuPlugin/EntityContextMenuPlugin.qml:52:3: QML RenderWindowOverlay: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[ign gazebo-1] [GUI] [Wrn] [Application.cc:753] [QT] file::/EntityContextMenuPlugin/EntityContextMenuPlugin.qml:67:3: QML EntityContextMenu: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[ign gazebo-1] [Msg] Ignition Gazebo GUI    v6.9.0
[ign gazebo-1] [Dbg] [Application.cc:90] Initializing application.
[ign gazebo-1] [GUI] [Dbg] [Application.cc:511] Create main window
[ign gazebo-1] [GUI] [Dbg] [PathManager.cc:66] Requesting resource paths through [/gazebo/resource_paths/get]
[ign gazebo-1] [GUI] [Dbg] [Gui.cc:156] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[ign gazebo-1] [GUI] [Dbg] [PathManager.cc:55] Received resource paths.
[ign gazebo-1] [GUI] [Dbg] [Gui.cc:215] Requesting GUI from [/world/empty_platform/gui/info]...
[ign gazebo-1] [GUI] [Dbg] [GuiRunner.cc:145] Requesting initial state from [/world/empty_platform/state]...
[ign gazebo-1] [GUI] [Msg] Loading config [/home/mbzirc/.ignition/gazebo/6/gui.config]
[ign gazebo-1] [GUI] [Dbg] [Application.cc:388] Loading plugin [MinimalScene]
[ign gazebo-1] [GUI] [Msg] Added plugin [3D View] to main window
[ign gazebo-1] [GUI] [Msg] Loaded plugin [MinimalScene] from path [/usr/lib/x86_64-linux-gnu/ign-gui-6/plugins/libMinimalScene.so]
[ign gazebo-1] [GUI] [Dbg] [Application.cc:388] Loading plugin [EntityContextMenuPlugin]
[ign gazebo-1] [GUI] [Msg] Added plugin [Entity Context Menu] to main window
[ign gazebo-1] [GUI] [Msg] Loaded plugin [EntityContextMenuPlugin] from path [/usr/lib/x86_64-linux-gnu/ign-gazebo-6/plugins/gui/libEntityContextMenuPlugin.so]
[ign gazebo-1] [GUI] [Dbg] [Application.cc:388] Loading plugin [GzSceneManager]
[ign gazebo-1] [GUI] [Msg] Added plugin [Scene Manager] to main window
[ign gazebo-1] [GUI] [Msg] Loaded plugin [GzSceneManager] from path [/usr/lib/x86_64-linux-gnu/ign-gazebo-6/plugins/gui/libGzSceneManager.so]
[ign gazebo-1] [GUI] [Dbg] [Application.cc:388] Loading plugin [InteractiveViewControl]
[... cut ...]
[ign gazebo-1] )
[ign gazebo-1] Exiting postupdate worker thread (5)
[ign gazebo-1] [Dbg] [SimulationRunner.cc:518] Exiting postupdate worker thread (4)
[ign gazebo-1] [Dbg] [Sensors.cc:375] SensorsPrivate::Stop
[ERROR] [ign gazebo-1]: process has died [pid 192591, exit code -2, cmd 'ign gazebo -v 4 -r coast.sdf --render-engine ogre --force-version 6'].
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT) again, ignoring...
[INFO] [parameter_bridge-2]: process has finished cleanly [pid 192593]

iche033, 2022-05-25:
@boushakiM, looks like your system (without docker) is also doing software rendering with `llvmpipe` and not using your GPU. Can you post the output of:

```
glxinfo | grep render
```

iche033, 2022-05-25:
looks like it's not using your nvidia GPU. A few things to try that I can think of:
1. do some sanity checks to make sure the nvidia graphics card is there and the module is loaded:
    ```
    # check for nvidia card in your system
    sudo lspci | grep -i nvidia
    # verify nvidia_* modules are loaded
    lsmod | grep nvidia
    ```
1. run  `sudo nvidia-settings` and make sure your nvidia GPU is in use and there are no errors reported
1. reinstall the nvidia drivers. Make sure to remove existing one, reboot, reinstall, then reboot again

boushakiM, 2022-05-25:
> looks like it's not using your nvidia GPU. A few things to try that I can think of:
> 
>     1. do some sanity checks to make sure the nvidia graphics card is there and the module is loaded:
>        ```
>        # check for nvidia card in your system
>        sudo lspci | grep -i nvidia
>        # verify nvidia_* modules are loaded
>        lsmod | grep nvidia
>        ```
> 
>     2. run  `sudo nvidia-settings` and make sure your nvidia GPU is in use and there are no errors reported
> 
>     3. reinstall the nvidia drivers. Make sure to remove existing one, reboot, reinstall, then reboot again

1.a)

mbzirc@mbzirc:~$ sudo lspci | grep -i nvidia
[sudo] password for mbzirc: 
01:00.0 VGA compatible controller: NVIDIA Corporation GP106BM [GeForce GTX 1060 Mobile 6GB] (rev a1)
01:00.1 Audio device: NVIDIA Corporation GP106 High Definition Audio Controller (rev a1)

1.b) 
mbzirc@mbzirc:~$ lsmod | grep nvidia
mbzirc@mbzirc:~$ 

2) 
mbzirc@mbzirc:~$ sudo nvidia-settings

ERROR: NVIDIA driver is not loaded

ERROR: Unable to load info from any available system

(nvidia-settings:28788): GLib-GObject-CRITICAL **: 21:47:27.807: g_object_unref: assertion 'G_IS_OBJECT (object)' failed
** Message: 21:47:27.810: PRIME: No offloading required. Abort
** Message: 21:47:27.810: PRIME: is it supported? no

PS: I have updated the drivers many times!

boushakiM, 2022-05-25:
Please see this: 

mbzirc@mbzirc:~$ ubuntu-drivers devices
== /sys/devices/pci0000:00/0000:00:01.0/0000:01:00.0 ==
modalias : pci:v000010DEd00001C60sv00001043sd00001800bc03sc00i00
vendor   : NVIDIA Corporation
model    : GP106BM [GeForce GTX 1060 Mobile 6GB]
driver   : nvidia-driver-510 - distro non-free recommended
driver   : nvidia-driver-418-server - distro non-free
driver   : nvidia-driver-470-server - distro non-free
driver   : nvidia-driver-390 - distro non-free
driver   : nvidia-driver-450-server - distro non-free
driver   : nvidia-driver-470 - distro non-free
driver   : nvidia-driver-510-server - distro non-free
driver   : xserver-xorg-video-nouveau - distro free builtin

mbzirc@mbzirc:~$ sudo ubuntu-drivers autoinstall
[sudo] password for mbzirc: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
mbzirc@mbzirc:~$ sudo apt install nvidia-driver-510
Reading package lists... Done
Building dependency tree       
Reading state information... Done
nvidia-driver-510 is already the newest version (510.73.05-0ubuntu0.20.04.1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

boushakiM, 2022-05-25:
Thank you so much, I have solved following:
1)sudo apt-get remove --purge '^nvidia-.*'
2)sudo apt-get install ubuntu-desktop
3)reboot
4)sudo ubuntu-drivers autoinstall
5)udo apt install nvidia-driver-510
6)Reboot
Wrking!
Many many thanks for your reactivity.

iche033, 2022-05-25:
awesome, great to hear!

closed 2022-05-25 by iche033