osrf/mbzirc #123: dockerhub img does not work

femust, 2022-05-26:
Hey,

so when I follow instructions from Docker setup
```
Docker setup
Docker images are available on Docker Hub: https://hub.docker.com/repository/docker/osrf/mbzirc

Pull the latest version of the docker image

docker pull osrf/mbzirc:mbzirc_sim_latest
Clone the repo and launch a Docker container from the image using the run.bash script. Note: requires nvidia-docker2

git clone https://github.com/osrf/mbzirc.git
cd mbzirc/docker
bash run.bash osrf/mbzirc:mbzirc_sim_latest  /bin/bash
```
and I want to run
```
source install/setup.bash
ros2 launch mbzirc_ros competition_local.launch.py ign_args:="-v 4 -r simple_demo.sdf"
```
I got the following error
```
developer@femust:~/mbzirc_ws$ ros2 launch mbzirc_ros competition_local.launch.py ign_args:="-v 4 -r simple_demo.sdf"
[INFO] [launch]: All log files can be found below /home/developer/.ros/log/2022-05-26-10-10-35-289914-femust-98
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ign gazebo-1]: process started with pid [100]
[INFO] [parameter_bridge-2]: process started with pid [102]
[parameter_bridge-2] [INFO] [1653552635.423359775] [ros_ign_bridge]: Creating IGN->ROS Bridge: [/mbzirc/score] (ignition.msgs.Float -> std_msgs/msg/Float32) (Lazy 0): 
[parameter_bridge-2] [INFO] [1653552635.423492625] [ros_ign_bridge]: Creating IGN->ROS Bridge: [/clock] (ignition.msgs.Clock -> rosgraph_msgs/msg/Clock) (Lazy 0): 
[parameter_bridge-2] [INFO] [1653552635.423538966] [ros_ign_bridge]: Creating IGN->ROS Bridge: [/mbzirc/run_clock] (ignition.msgs.Clock -> rosgraph_msgs/msg/Clock) (Lazy 0): 
[parameter_bridge-2] [INFO] [1653552635.423552107] [ros_ign_bridge]: Creating IGN->ROS Bridge: [/mbzirc/phase] (ignition.msgs.StringMsg -> std_msgs/msg/String) (Lazy 0): 
[parameter_bridge-2] [INFO] [1653552635.423593648] [ros_ign_bridge]: Creating IGN->ROS Bridge: [/mbzirc/target/stream/status] (ignition.msgs.StringMsg -> std_msgs/msg/String) (Lazy 0): 
[ign gazebo-1] No protocol specified
[ign gazebo-1] qt.qpa.xcb: could not connect to display :0
[ign gazebo-1] qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
[ign gazebo-1] This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.
[ign gazebo-1] 
[ign gazebo-1] Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, xcb.
[ign gazebo-1] 
[ign gazebo-1] Stack trace (most recent call last):
[ign gazebo-1] #31   Object "/usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f94aba412ed, in ruby_run_node
[ign gazebo-1] #30   Object "/usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f94aba3c490, in 
[ign gazebo-1] #29   Object "/usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f94abbab86f, in rb_vm_exec
[ign gazebo-1] #28   Object "/usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f94abba5130, in 
[ign gazebo-1] #27   Object "/usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f94abb94405, in 
[ign gazebo-1] #26   Object "/usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f94abbaddaa, in 
[ign gazebo-1] #25   Object "/usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f94abba1025, in 
[ign gazebo-1] #24   Object "/usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f94abaff1be, in 
[ign gazebo-1] #23   Object "/usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f94aba3f5d2, in rb_protect
[ign gazebo-1] #22   Object "/usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f94abbb89f0, in rb_yield
[ign gazebo-1] #21   Object "/usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f94abbab86f, in rb_vm_exec
[ign gazebo-1] #20   Object "/usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f94abba5130, in 
[ign gazebo-1] #19   Object "/usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f94abb94405, in 
[ign gazebo-1] #18   Object "/usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f94abbaddaa, in 
[ign gazebo-1] #17   Object "/usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f94abba1025, in 
[ign gazebo-1] #16   Object "/usr/lib/x86_64-linux-gnu/ruby/2.7.0/fiddle.so", at 0x7f94a7f81714, in 
[ign gazebo-1] #15   Object "/usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f94abb726d9, in rb_nogvl
[ign gazebo-1] #14   Object "/usr/lib/x86_64-linux-gnu/ruby/2.7.0/fiddle.so", at 0x7f94a7f818fb, in 
[ign gazebo-1] #13   Object "/usr/lib/x86_64-linux-gnu/libffi.so.7", at 0x7f94a7f76409, in 
[ign gazebo-1] #12   Object "/usr/lib/x86_64-linux-gnu/libffi.so.7", at 0x7f94a7f76ff4, in 
[ign gazebo-1] #11   Object "/usr/lib/x86_64-linux-gnu/libignition-gazebo6-ign.so.6.9.0", at 0x7f94a750f740, in runGui
[ign gazebo-1] #10   Object "/usr/lib/x86_64-linux-gnu/libignition-gazebo6-gui.so.6", at 0x7f94a73143cf, in ignition::gazebo::v6::gui::runGui(int&, char**, char const*, char const*)
[ign gazebo-1] #9    Object "/usr/lib/x86_64-linux-gnu/libignition-gazebo6-gui.so.6", at 0x7f94a73125ae, in ignition::gazebo::v6::gui::createGui(int&, char**, char const*, char const*, bool, char const*)
[ign gazebo-1] #8    Object "/usr/lib/x86_64-linux-gnu/libignition-gui6.so.6", at 0x7f94a5d7ff57, in ignition::gui::Application::Application(int&, char**, ignition::gui::WindowType)
[ign gazebo-1] #7    Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f94a581d3bc, in QApplicationPrivate::init()
[ign gazebo-1] #6    Object "/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5", at 0x7f94a482a542, in QGuiApplicationPrivate::init()
[ign gazebo-1] #5    Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f94a6079f54, in QCoreApplicationPrivate::init()
[ign gazebo-1] #4    Object "/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5", at 0x7f94a4828707, in QGuiApplicationPrivate::createEventDispatcher()
[ign gazebo-1] #3    Object "/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5", at 0x7f94a48277ad, in QGuiApplicationPrivate::createPlatformIntegration()
[ign gazebo-1] #2    Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f94a5e72aac, in QMessageLogger::fatal(char const*, ...) const
[ign gazebo-1] #1    Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7f94ab7c3858, in abort
[ign gazebo-1] #0    Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7f94ab7e400b, in gsignal
[ign gazebo-1] Aborted (Signal sent by tkill() 121 1001)
[ign gazebo-1] [Wrn] [LocalCache.cc:105] Server directory does not exist [/home/developer/.ignition/fuel/fuel.ignitionrobotics.org]
[ign gazebo-1] Escalating to SIGKILL on [Ignition Gazebo Server]
[INFO] [ign gazebo-1]: process has finished cleanly [pid 100]
^[[A^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[parameter_bridge-2] [INFO] [1653552704.152304698] [rclcpp]: signal_handler(signal_value=2)
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT) again, ignoring...
[INFO] [parameter_bridge-2]: process has finished cleanly [pid 102]

```
I wonder what's the reason why the dockerhub image is falling, while if I build the image from scratch on my PC like this: 
```
To build a docker image of the simulator locally:

Navigate to the docker directory and build the mbzirc_sim Docker image

cd mbzirc
bash docker/build.bash mbzirc_sim
The process can take a few minutes. Once it is done, you can launch the Docker container:

bash run.bash mbzirc_sim
```
it works. Any thoughts?

iche033, 2022-05-26:
hi, we tested on a couple of machines and coudn't reproduce the issue. The images from dockerhub are working for us. What's your host system? Ubuntu? Do you have an X server running?

femust, 2022-05-27:
Ubuntu 20.04 Classic setup, nothing fancy, but on pc when built from scratch and built via docker it works (though i thought I would use the image for a docker and code development across the team)