osrf/mbzirc #147: Image publishing will  slow down to 3hz while we use `ros2 topic hz` or `ros2 bag record` command, and computer resources are not used up.

xirhxq, 2022-06-14:
[image]

iche033, 2022-06-15:
What does the GPU resource / VRAM usage look like? I notice that everything slows down if VRAM is filled up. If you have an nvidia card, you can run `nvidia-smi` to get some info on GPU usage

xirhxq, 2022-06-16:
[image]

xirhxq, 2022-06-16:
> What does the GPU resource / VRAM usage look like? I notice that everything slows down if VRAM is filled up. If you have an nvidia card, you can run `nvidia-smi` to get some info on GPU usage

After running `nvidia-smi` command and 'ros2 topic hz' command, it looks like this:
[image]

I can also notice this problem when running `ros2 run rqt_image_view rqt_image_view` command to see the image from the camera on spawned UAV. Without `rostopic hz` the fps seems normal, but seems frozen with that command.

iche033, 2022-06-17:
I suspect that `ros2 topic hz` is not using sim time. I wrote a simple `subscriber.py` to verify the timestamps of the image topics:

You can change ` /quadrotor_1/slot0/image_raw` in the code below to your topic name and run `python3 subscriber.py`

<details><summary>subscriber.py</summary>

```py
import rclpy
from rclpy.node import Node

from std_msgs.msg import String
from sensor_msgs.msg import Image

class MinimalSubscriber(Node):

    def __init__(self):
        super().__init__('minimal_subscriber')
        self.subscription = self.create_subscription(
            Image,
            '/quadrotor_1/slot0/image_raw',
            self.listener_callback,
            10)
        self.subscription  # prevent unused variable warning

    def listener_callback(self, msg):
        print(f'Timestamp {msg.header.stamp.sec} {msg.header.stamp.nanosec}')

def main(args=None):
    rclpy.init(args=args)

    minimal_subscriber = MinimalSubscriber()

    rclpy.spin(minimal_subscriber)

    # Destroy the node explicitly
    # (optional - otherwise it will be done automatically
    # when the garbage collector destroys the node object)
    minimal_subscriber.destroy_node()
    rclpy.shutdown()

if __name__ == '__main__':
    main()
```

</details>

I'm testing with an HD camera and the timestamps looks correct at ~20Hz on my end. Can you give that a try and see if the timestamps are correct in your setup?

xirhxq, 2022-06-17:
Thanks for your effort! The timestamps also look fine on my end. So I am guessing there really are some unknown problems in `ros2 ropic hz`.

[image]

However there is still something making me puzzled: when I have spawned an UAV with an HD camera in the coast environment on the ground and adjust its height (z value) directly in the ignition interface, I noticed some difference between   `rqt_image_view` and ignition interface with `ros2 topic hz` command

- In Ignition interface, the UAV immediately dropped on to the ground;
- In `rqt_image_view`, the image of the UAV seems frozen, like the UAV didn't leave the ground at all

The situation looks like below:

https://user-images.githubusercontent.com/50697317/174224523-878e9ea0-bf19-4fbb-8555-75f09dd3cb27.mp4

xirhxq, 2022-06-17:
https://user-images.githubusercontent.com/50697317/174224690-4ae5ab5c-fa32-4d84-944c-a217f0287e01.mp4

The video looks corrupt on the last repost so I re-upload it.

iche033, 2022-06-17:
I have not be able to reproduce this yet. From looking at the video, it seems like `rqt_image_view` does not display images if `ros2 topic hz` is running, and after you killed `ros2 topic hz`, the images started coming through? I wonder if it's caused by multiple subscriptions. One thing to try is to open multiple `rqt_image_view` and subscribe to the same topic and see if the issue also occurs.

xirhxq, 2022-06-19:
I've tried multiple `rqt_image_view` and the issue still occurred. It seems like `rqt_image_view` & `ros2 topic hz` subscribing the same topic would cause this problem. So it is anything I can do to avoid that in ros2 environment?

Also, inspired by this: 

> I suspect that `ros2 topic hz` is not using sim time. 

we 

1. used `ros2 run ros1_bridge dynamic_bridge --bridge-all-topics` to bridge all topics from ros2 to ros1;
2. set the local time of ros1 environment to sim time, using `rosparam set use_sim_time True`;
3. proofed that the ros1 time `ros::time::now()` and sim time `/clock` are equivalent;

then by using `rostopic hz` to see the frequency of camera, lidar & imu data. Things are fine with camera & imu, which frequency is `30hz` & `250hz`, however after a while the frequency of `lidar` topic dropped from `30hz` to `8hz`. I wonder if it is because the data is accumulating on and on. Could I see the original settings of topic frequency of such sensors?