osrf/mbzirc #152: Switching from setup to run phase manually.

fzoric8, 2022-06-17:
Hi, 

I'm trying to test some subparts of a system, and `coast.sdf` is computationally to demanding for my system. 

Therefore, I'm trying to do following: 
* spawn UAVs and objects in `empty.world` 
* recreate competition functionality required for competition (GameLogic plugin etc...) 

After spawning UAVs and flying for a bit (30 secs). Everything stops. 

My UAVs get `uav_namespace_static_link` element in simulation and I'm not able to continue movement. 

Output of simulator is: 
```
[ign gazebo-1] [Msg] Scoring has Started

```

I was looking at [this](https://github.com/osrf/mbzirc/wiki/Competition-Tasks-and-Time-Penalties), but I'm not 
couldn't notice any `ign service` that enables me to start `run` phase. 

**I'm looking for a command or service that enables me switching from setup to starting of a simulation.** 

Thank you in advance, 
Best regards.

iche033, 2022-06-17:
Hi, 

> My UAVs get uav_namespace_static_link element in simulation and I'm not able to continue movement.

This looks like the UAV went out of bounds. Feel free to increase the size of the geofence in your test world.

> [ign gazebo-1] [Msg] Scoring has Started

This indicates that simulation phase should have switched from `setup` to `started`. This happens when a) a vehicle spawned in simulation has moved (for ~5m) from its original position, or b) the sim time elapsed has exceeded the allowed set up time specified in `<setup_duration_seconds>` in the GameLogicPlugin.

> I'm looking for a command or service that enables me switching from setup to starting of a simulation.

You can manually move your vehicle to trigger the change in phase or use the command:

```
ign service -s /mbzirc/skip_to_phase --reqtype ignition.msgs.StringMsg --reptype ignition.msgs.Boolean --timeout 300 --req 'data: "started"
```

fzoric8, 2022-06-27:
Thank you very much, this solved my problem.
Closing the issue.

closed 2022-06-27 by fzoric8