osrf/mbzirc #154: Accessing slots from different robots

zorawar12, 2022-06-19:
Hello sir,

I am trying to connect two Quadcopters using a chain model that I have created. I want to connect the chain just like any other sensor through the exposed slots. 

However, one needs to specify the slot components while launching the robot. Can you please tell me how to launch two UAVs with their slots sharing the same sensor (In my case it would rather be a chain) object?

Thank you

iche033, 2022-06-21:
Our launch script dynamically generate the vehicle's model.sdf file based on the sensors specified and the sensor becomes a part of the vehicle model, i.e. one single model. So two vehicles won't be able to share the same chain model. 

I just want to note that for final evaluation, teams won't be able to use custom models / sensors. But for local testing, you're welcome to modify the sim and use any custom models you like.

One possible approach to achieve what you are looking to do would be to use a DetachableJoint plugin which create a joint between two models. Here is an example:
https://github.com/osrf/mbzirc/blob/main/mbzirc_ign/worlds/coast.sdf#L1138-L1143
I imagine for your use case, you would have a quadrotor with the chain model in one slot and it would have the Detachable joint plugin that creates a joint with another quadrotor. However, I can't guarantee that will work well as the physics simulation may become unstable. One likely result is that the quadrotor won't be able to take off and would require tuning some physics params.

zorawar12, 2022-06-22:
Thank you for the information. I will try in my free time this interesting stuff. Thanks for sharing.

closed 2022-07-02 by zorawar12