osrf/mbzirc #165: Gazebo crash during physical interaction

fzoric8, 2022-06-27:
When trying to grasp the object using aerial manipulator (small or large) I quite often experience simulation crash with following output: 

```
[ign gazebo-1] 
[ign gazebo-1] ODE INTERNAL ERROR 1: assertion "aabbBound >= dMinIntExact && aabbBound < dMaxIntExact" failed in collide() [collision_space.cpp:460]
[ign gazebo-1] Stack trace (most recent call last):
[ign gazebo-1] #31   Object "/usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f91e0002daa, in 
[ign gazebo-1] #30   Object "/usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f91dfff6025, in 
[ign gazebo-1] #29   Object "/usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f91dff541be, in 
[ign gazebo-1] #28   Object "/usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f91dfe945d2, in rb_protect
[ign gazebo-1] #27   Object "/usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f91e000d9f0, in rb_yield
[ign gazebo-1] #26   Object "/usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f91e000086f, in rb_vm_exec
[ign gazebo-1] #25   Object "/usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f91dfffa130, in 
[ign gazebo-1] #24   Object "/usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f91dffe9405, in 
[ign gazebo-1] #23   Object "/usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f91e0002daa, in 
[ign gazebo-1] #22   Object "/usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f91dfff6025, in 
[ign gazebo-1] #21   Object "/usr/lib/x86_64-linux-gnu/ruby/2.7.0/fiddle.so", at 0x7f91dc3d6714, in 
[ign gazebo-1] #20   Object "/usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f91dffc76d9, in rb_nogvl
[ign gazebo-1] #19   Object "/usr/lib/x86_64-linux-gnu/ruby/2.7.0/fiddle.so", at 0x7f91dc3d68fb, in 
[ign gazebo-1] #18   Object "/usr/lib/x86_64-linux-gnu/libffi.so.7", at 0x7f91dc3cb409, in 
[ign gazebo-1] #17   Object "/usr/lib/x86_64-linux-gnu/libffi.so.7", at 0x7f91dc3cbff4, in 
[ign gazebo-1] #16   Object "/usr/lib/x86_64-linux-gnu/libignition-gazebo6-ign.so.6.8.0", at 0x7f91db950f65, in runServer
[ign gazebo-1] #15   Object "/usr/lib/x86_64-linux-gnu/libignition-gazebo6.so.6", at 0x7f91db56f81b, in ignition::gazebo::v6::Server::Run(bool, unsigned long, bool)
[ign gazebo-1] #14   Object "/usr/lib/x86_64-linux-gnu/libignition-gazebo6.so.6", at 0x7f91db578713, in 
[ign gazebo-1] #13   Object "/usr/lib/x86_64-linux-gnu/libignition-gazebo6.so.6", at 0x7f91db586d32, in ignition::gazebo::v6::SimulationRunner::Run(unsigned long)
[ign gazebo-1] #12   Object "/usr/lib/x86_64-linux-gnu/libignition-gazebo6.so.6", at 0x7f91db5864c4, in ignition::gazebo::v6::SimulationRunner::Step(ignition::gazebo::v6::UpdateInfo const&)
[ign gazebo-1] #11   Object "/usr/lib/x86_64-linux-gnu/libignition-gazebo6.so.6", at 0x7f91db57ed71, in ignition::gazebo::v6::SimulationRunner::UpdateSystems()
[ign gazebo-1] #10   Object "/usr/lib/x86_64-linux-gnu/ign-gazebo-6/plugins/libignition-gazebo-physics-system.so", at 0x7f91cc50e625, in ignition::gazebo::v6::systems::Physics::Update(ignition::gazebo::v6::UpdateInfo const&, ignition::gazebo::v6::EntityComponentManager&)
[ign gazebo-1] #9    Object "/usr/lib/x86_64-linux-gnu/ign-gazebo-6/plugins/libignition-gazebo-physics-system.so", at 0x7f91cc4f962c, in ignition::gazebo::v6::systems::PhysicsPrivate::Step(std::chrono::duration<long, std::ratio<1l, 1000000000l> > const&)
[ign gazebo-1] #8    Object "/usr/lib/x86_64-linux-gnu/ign-physics-5/engine-plugins/libignition-physics-dartsim-plugin.so", at 0x7f91bc922d5d, in ignition::physics::dartsim::SimulationFeatures::WorldForwardStep(ignition::physics::Identity const&, ignition::physics::SpecifyData<ignition::physics::RequireData<ignition::physics::WorldPoses>, ignition::physics::ExpectData<ignition::physics::ChangedWorldPoses, ignition::physics::Contacts, ignition::physics::JointPositions> >&, ignition::physics::CompositeData&, ignition::physics::ExpectData<ignition::physics::ApplyExternalForceTorques, ignition::physics::ApplyGeneralizedForces, ignition::physics::VelocityControlCommands, ignition::physics::ServoControlCommands> const&)
[ign gazebo-1] #7    Object "/usr/lib/x86_64-linux-gnu/libdart.so.6", at 0x7f91bc5f3d20, in dart::simulation::World::step(bool)
[ign gazebo-1] #6    Object "/usr/lib/x86_64-linux-gnu/libdart.so.6", at 0x7f91bc5d87b5, in dart::constraint::ConstraintSolver::solve()
[ign gazebo-1] #5    Object "/usr/lib/x86_64-linux-gnu/libdart.so.6", at 0x7f91bc5d727d, in dart::constraint::ConstraintSolver::updateConstraints()
[ign gazebo-1] #4    Object "/usr/lib/x86_64-linux-gnu/libdart-collision-ode.so.6", at 0x7f91bc741b6b, in dart::collision::OdeCollisionDetector::collide(dart::collision::CollisionGroup*, dart::collision::CollisionOption const&, dart::collision::CollisionResult*)
[ign gazebo-1] #3    Object "/usr/lib/x86_64-linux-gnu/libode.so.8", at 0x7f91b72d2456, in dxHashSpace::collide(void*, void (*)(void*, dxGeom*, dxGeom*))
[ign gazebo-1] #2    Object "/usr/lib/x86_64-linux-gnu/libode.so.8", at 0x7f91b72dac57, in dDebug
[ign gazebo-1] #1    Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7f91dfc18858, in abort
[ign gazebo-1] #0    Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7f91dfc3903b, in gsignal
[ign gazebo-1] Aborted (Signal sent by tkill() 78697 1000)
[ign gazebo-1] Stack trace (most recent call last):
[ign gazebo-1] #31   Object "/usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f91dfe91490, in 
[ign gazebo-1] #30   Object "/usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f91e000086f, in rb_vm_exec
[ign gazebo-1] #29   Object "/usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f91dfffa130, in 
[ign gazebo-1] #28   Object "/usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f91dffe9405, in 
[ign gazebo-1] #27   Object "/usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f91e0002daa, in 
[ign gazebo-1] #26   Object "/usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f91dfff6025, in 
[ign gazebo-1] #25   Object "/usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f91dff541be, in 
[ign gazebo-1] #24   Object "/usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f91dfe945d2, in rb_protect
[ign gazebo-1] #23   Object "/usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f91e000d9f0, in rb_yield
[ign gazebo-1] #22   Object "/usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f91e000086f, in rb_vm_exec
[ign gazebo-1] #21   Object "/usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f91dfffa130, in 
[ign gazebo-1] #20   Object "/usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f91dffe9405, in 
[ign gazebo-1] #19   Object "/usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f91e0002daa, in 
[ign gazebo-1] #18   Object "/usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f91dfff6025, in 
[ign gazebo-1] #17   Object "/usr/lib/x86_64-linux-gnu/ruby/2.7.0/fiddle.so", at 0x7f91dc3d6714, in 
[ign gazebo-1] #16   Object "/usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f91dffc76d9, in rb_nogvl
[ign gazebo-1] #15   Object "/usr/lib/x86_64-linux-gnu/ruby/2.7.0/fiddle.so", at 0x7f91dc3d68fb, in 
[ign gazebo-1] #14   Object "/usr/lib/x86_64-linux-gnu/libffi.so.7", at 0x7f91dc3cb409, in 
[ign gazebo-1] #13   Object "/usr/lib/x86_64-linux-gnu/libffi.so.7", at 0x7f91dc3cbff4, in 
[ign gazebo-1] #12   Object "/usr/lib/x86_64-linux-gnu/libignition-gazebo6-ign.so.6.8.0", at 0x7f91db950780, in runGui
[ign gazebo-1] #11   Object "/usr/lib/x86_64-linux-gnu/libignition-gazebo6-gui.so.6", at 0x7f91db744468, in ignition::gazebo::v6::gui::runGui(int&, char**, char const*, char const*)
[ign gazebo-1] #10   Object "/usr/lib/x86_64-linux-gnu/libignition-gui6.so.6", at 0x7f91da136adc, in ignition::gui::Application::~Application()
[ign gazebo-1] #9    Object "/usr/lib/x86_64-linux-gnu/libignition-gui6.so.6", at 0x7f91da136836, in ignition::gui::Application::~Application()
[ign gazebo-1] #8    Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f91d9bcc49d, in QApplication::~QApplication()
[ign gazebo-1] #7    Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f91da42a97d, in QCoreApplication::~QCoreApplication()
[ign gazebo-1] #6    Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f91da45c4be, in QObject::~QObject()
[ign gazebo-1] #5    Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f91da451eed, in QObjectPrivate::deleteChildren()
[ign gazebo-1] #4    Object "/usr/lib/x86_64-linux-gnu/libignition-gazebo6-gui.so.6", at 0x7f91db74e39c, in ignition::gazebo::v6::GuiRunner::~GuiRunner()
[... cut ...]  ignition::gazebo::v6::detail::ComponentTypeHasher, std::__detail::_Mod_range_hashing, std::__detail::_Default_ranged_hash, std::__detail::_Prime_rehash_policy, std::__detail::_Hashtable_traits<true, false, true> >::clear()
[ign gazebo-1] Segmentation fault (Address not mapped to object [0x7f91b5edd048])
[INFO] [ign gazebo-1]: process has finished cleanly [pid 78624]
```
I've encountered it multiple times and I'm quite sure it's not related to my program. 

I find no clear reason why it happens.

iche033, 2022-06-28:
Does this happen with the oberon7 finger gripper? yeah a video will help to see how it happens. We'll try to reproduce it.

fzoric8, 2022-06-29:
Yes, with oberon7 finger gripper. 

However, here is the [video](https://streamable.com/btev3y). 

Most often it crashes during physical interaction, however sometimes it crashes as you can see from the video. Only this UAV in video is moving.

fzoric8, 2022-07-06:
Any updates on this?  

Tnx

iche033, 2022-07-06:
I did some tests with the same setup involving a hexrotor with oberon7 gripper and small dry bag but haven't been able to reproduce it yet. 

There are a few things to double check:
* I'm doing my tests in the simple_demo world. I noticed that you have created a test environment. Can you check if you have the same physics [max_step_size](https://github.com/osrf/mbzirc/blob/v4/mbzirc_ign/worlds/simple_demo.sdf#L7) in your setup as the simple_demo / coast world?
* Are you using the latest release or latest `main` branch? In our latest release, we removed tri-meshes from the oberon7 gripper model which may help improve physics stability and performance
* Is your small dry bag model the [latest version](https://app.gazebosim.org/OpenRobotics/fuel/models/Small%20Dry%20Bag) (version 3). You can check the local cache `~/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/small\ dry\ bag/{version number}`. If not, you can remove the whole directory and the simulator will download the latest one.

We'll keep testing.

fzoric8, 2022-08-22:
Problem persists on another PC: 

Error output is: 
```
[ign gazebo-1] ODE INTERNAL ERROR 1: assertion "aabbBound >= dMinIntExact && aabbBound < dMaxIntExact" failed in collide() [collision_space.cpp:460]
```

iche033, 2022-08-23:
so far, we haven't been able to reproduce the issue. However, I have seen these types of errors before in other simulations during object collisions. We will make a note this and keep an eye out for these errors when during dry-run and final evaluation. If we do see that simulator crashed because of this issue, we'll re-run the simulation.

closed 2023-02-01 by fzoric8