osrf/mbzirc #175: Robotic arm is constrained to rotate towards the USV

m00nlab, 2022-07-04:
Hello sir,

I want to ask that the current model that has been provided for the robotic arm has its azimuth constrained to rotate within the limit of (-1.04719758, 1.04719758 ). However, as part of the mission, we need to unload the payload on the USV. Is there any future plans towards changing the designs? Or we are allowed to change the Base?

iche033, 2022-07-06:
hi, if you update to the latest version, it should have increased [azimuth joint limits (± 2.0944 rad)](https://github.com/osrf/mbzirc/wiki/Robot-and-sensor-APIs#manipulator).

m00nlab, 2022-07-06:
> hi, if you update to the latest version, it should have increased [azimuth joint limits (± 2.0944 rad)](https://github.com/osrf/mbzirc/wiki/Robot-and-sensor-APIs#manipulator).

Yes, true, I see that, however, the model has two screws at the base, so mechanically that should obstruct the arm from rotating further, so after (-1.04719758, 1.04719758 ), the model should self collide. I hope in the final evaluation, that won't be a problem.

iche033, 2022-07-06:
the base collisions should already have been adjusted so that there would not be any self-collisions. This will remain the same in the final evaluation in simulation phase:

[image]

Let me know if you still have issues with limits.

In the physical competition, I believe you can pick any arm you like to use and not limited to the Oberon / Schilling arm.

m00nlab, 2022-07-09:
Thank you, that clarifies my query!

closed 2022-07-09 by m00nlab