osrf/mbzirc #183: IP settings problem in docker compose

m00nlab, 2022-07-20:
Hello,

I ran the generated docker compose file. However, I am not being able to see the simulator display in gui mode. Here is the output of the docker compose: With run.bash, the display shows up but with compose, it does not.

>>[+] Running 5/5
[/mbzirc/score (ignition.msgs.Float) -> /mbzirc/score (std_msgs/msg/Float32)] (Lazy 0)
compose-bridge1-1    | [parameter_bridge-6] [INFO] [1658343218.140491380] [ros_ign_bridge]: Creating IGN->ROS Bridge: [/clock (ignition.msgs.Clock) -> /clock (rosgraph_msgs/msg/Clock)] (Lazy 0)
compose-bridge1-1    | [parameter_bridge-6] [INFO] [1658343218.141054363] [ros_ign_bridge]: Creating IGN->ROS Bridge: [/mbzirc/run_clock (ignition.msgs.Clock) -> /mbzirc/run_clock (rosgraph_msgs/msg/Clock)] (Lazy 0)
compose-bridge1-1    | [parameter_bridge-6] [INFO] [1658343218.141348718] [ros_ign_bridge]: Creating IGN->ROS Bridge: [/mbzirc/phase (ignition.msgs.StringMsg) -> /mbzirc/phase (std_msgs/msg/String)] (Lazy 0)
compose-bridge1-1    | [parameter_bridge-6] [INFO] [1658343218.141804397] [ros_ign_bridge]: Creating IGN->ROS Bridge: [/mbzirc/target/stream/status (ignition.msgs.StringMsg) -> /mbzirc/target/stream/status (std_msgs/msg/String)] (Lazy 0)
compose-solution1-1  | [INFO] [launch]: All log files can be found below /tmp/ign/logs/ros/2022-07-20-18-53-38-592974-afbf94c6da5a-55
compose-solution1-1  | [INFO] [launch]: Default logging verbosity is set to INFO

compose-sim-1        | [ign gazebo-1] [GUI] [Dbg] [GuiRunner.cc:375] Loaded system [ignition::gazebo::systems::WaveVisual] for entity [12] in GUI
compose-sim-1        | [ign gazebo-1] [GUI] [Msg] Loading plugin [ignition-rendering-ogre2]
compose-sim-1        | [ign gazebo-1] [GUI] [Dbg] [MinimalScene.cc:534] Create scene [scene]
compose-sim-1        | [ign gazebo-1] [GUI] [Msg] Loading plugin [ignition-rendering-ogre2]
compose-sim-1        | [ign gazebo-1] [GUI] [Dbg] [ViewAngle.cc:346] ViewAngle plugin is moving camera [scene::Camera(65527)]
compose-sim-1        | [ign gazebo-1] [GUI] [Msg] Loading plugin [ignition-rendering-ogre2]
compose-sim-1        | [ign gazebo-1] [GUI] [Dbg] [VideoRecorder.cc:151] Video Recorder plugin is recoding camera [scene::Camera(65527)]
compose-sim-1        | [ign gazebo-1] [GUI] [Msg] Video recorder stats topic advertised on [/gui/record_video/stats]
compose-sim-1        | [ign gazebo-1] [GUI] [Msg] Loading plugin [ignition-rendering-ogre2]
compose-sim-1        | [ign gazebo-1] [GUI] [Dbg] [TransformControl.cc:528] TransformControl plugin is using camera [scene::Camera(65527)]
compose-sim-1        | [ign gazebo-1] [GUI] [Msg] Loading plugin [ignition-rendering-ogre2]
compose-sim-1        | [ign gazebo-1] [GUI] [Dbg] [Spawn.cc:289] Spawn plugin is using camera [scene::Camera(65527)]
compose-sim-1        | [ign gazebo-1] [GUI] [Msg] Loading plugin [ignition-rendering-ogre2]
compose-sim-1        | [ign gazebo-1] [Wrn] [Ogre2RenderTarget.cc:575] Anti-aliasing level of '2' is not supported; valid FSAA levels are: [ 0 4 ]. Setting to 0
compose-sim-1        | [ign gazebo-1] [GUI] [Msg] Loading plugin [
compose-sim-1        | [ign gazebo-1] [Dbg] [SimulationRunner.cc:502] Creating postupdate worker thread (21)
compose-sim-1        | [ign gazebo-1] [Dbg] [SimulationRunner.cc:502] Creating postupdate worker thread (22)
compose-sim-1        | [ign gazebo-1] [Dbg] [SimulationRunner.cc:502] Creating postupdate worker thread (23)
compose-sim-1        | [ign gazebo-1] [Msg] Address [quadrotor_2] bound to model [quadrotor_2] on topic [model/quadrotor_2/rx]
compose-sim-1        | [ign gazebo-1] [Dbg] [CommsEndpoint.cc:89] Succesfuly bound to [quadrotor_2] on topic [model/quadrotor_2/rx]
compose-sim-1        | [ign gazebo-1] [Msg] Address [usv] bound to model [usv] on topic [model/usv/rx]
compose-sim-1        | [ign gazebo-1] [Dbg] [CommsEndpoint.cc:89] Succesfuly bound to [usv] on topic [model/usv/rx]
compose-sim-1        | [ign gazebo-1] [Msg] Address [quadrotor_1] bound to model [quadrotor_1] on topic [model/quadrotor_1/rx]
compose-sim-1        | [ign gazebo-1] [Dbg] [CommsEndpoint.cc:89] Succesfuly bound to [quadrotor_1] on topic [model/quadrotor_1/rx]
compose-sim-1        | [ign gazebo-1] [Dbg] [RenderUtil.cc:2529] Create scene [scene]
compose-sim-1        | [ign gazebo-1] [Dbg] [Sensors.cc:254] Rendering Thread initialized
compose-sim-1        | [ign gazebo-1] [Dbg] [Ogre2ParticleEmitter.cc:514] Particle emitter initialized
compose-sim-1        | [ign gazebo-1] [Dbg] [Ogre2ParticleEmitter.cc:514] Particle emitter initialized
compose-sim-1        | [ign gazebo-1] [Dbg] [Ogre2ParticleEmitter.cc:514] Particle emitter initialized
compose-sim-1        | [ign gazebo-1] [Dbg] [Ogre2ParticleEmitter.cc:514] Particle emitter initialized
compose-sim-1        | [ign gazebo-1] [Dbg] [Ogre2ParticleEmitter.cc:514] Particle emitter initialized
compose-sim-1        | [ign gazebo-1] [Dbg] [Ogre2ParticleEmitter.cc:514] Particle emitter initialized
compose-sim-1        | [ign gazebo-1] [Dbg] [CameraSensor.cc:297] Camera images for [quadrotor_2::sensor_0::sensor_link::camera] advertised on [world/coast/model/quadrotor_2/model/sensor_0/link/sensor_link/sensor/camera/image]
compose-sim-1        | [ign gazebo-1] [Dbg] [CameraSensor.cc:598] Camera info for [quadrotor_2::sensor_0::sensor_link::camera] advertised on [/world/coast/model/quadrotor_2/model/sensor_0/link/sensor_link/sensor/camera/camera_info]
compose-sim-1        | [ign gazebo-1] [Dbg] [CameraSensor.cc:297] Camera images for [usv::sensor_0::sensor_link::camera] advertised on [world/coast/model/usv/model/sensor_0/link/sensor_link/sensor/camera/image]
compose-sim-1        | [ign gazebo-1] [Dbg] [CameraSensor.cc:598] Camera info for [usv::sensor_0::sensor_link::camera] advertised on [/world/coast/model/usv/model/sensor_0/link/sensor_link/sensor/camera/camera_info]
compose-sim-1        | [ign gazebo-1] [Dbg] [CameraSensor.cc:297] Camera images for [quadrotor_1::sensor_0::sensor_link::camera] advertised on [world/coast/model/quadrotor_1/model/sensor_0/link/sensor_link/sensor/camera/image]
compose-sim-1        | [ign gazebo-1] [Dbg] [CameraSensor.cc:598] Camera info for [quadrotor_1::sensor_0::sensor_link::camera] advertised on [/world/coast/model/quadrotor_1/model/sensor_0/link/sensor_link/sensor/camera/camera_info]
compose-sim-1        | [ign gazebo-1] [Msg] Loading plugin [ignition-rendering-ogre2]
compose-sim-1        | [ign gazebo-1] [Dbg] [CameraSensor.cc:415] Enabling camera sensor: 'usv::sensor_0::sensor_link::camera' data generation.

iche033, 2022-07-21:
hmm the attached output does not reveal the error. The error could be printed out earlier in the terminal. Are you able to attach the whole output? You can run `docker compose -f mbzirc_compose.yaml up` without `-d` arg to see the console msgs. My current guess is that rendering failed to start due to X auth issue in the simulation docker container started by docker-compose.

m00nlab, 2022-07-21:
The error message is too long to post. Here are some suspicious lines:

>[GUI] [Wrn] [Application.cc:753] [QT] QXcbConnection: XCB error: 3 (BadWindow), sequence: 179, resource id: 6291470, major code: 20 (GetProperty), minor code: 0
compose-sim-1        | [ign gazebo-1] [GUI] [Wrn] [Application.cc:753] [QT] QXcbConnection: XCB error: 3 (BadWindow), sequence: 180, resource id: 6291470, major code: 20 (GetProperty), minor code: 0
compose-sim-1        | [ign gazebo-1] libEGL warning: failed to open /dev/dri/renderD128: Permission denied
compose-sim-1        | [ign gazebo-1] 
compose-sim-1        | [ign gazebo-1] libEGL warning: failed to open /dev/dri/renderD128: Permission denied
compose-sim-1        | [ign gazebo-1] 
compose-sim-1  d 

I ran it with the run.bash, it worked fine....I could see the display

m00nlab, 2022-07-21:
If I undo everything I did, following is the error I am getting:

>compose-sim-1        | [ign gazebo-1] [GUI] [Err] [Ogre2RenderEngine.cc:973]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): OpenGL 3.3 is not supported. Please update your graphics card drivers. in GL3PlusRenderSystem::initialiseContext at /var/lib/jenkins/workspace/ogre-2.2-debbuilder/repo/RenderSystems/GL3Plus/src/OgreGL3PlusRenderSystem.cpp (line 3364)
compose-sim-1        | [ign gazebo-1] [GUI] [Err] [Ogre2RenderEngine.cc:973]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): OpenGL 3.3 is not supported. Please update your graphics card drivers. in GL3PlusRenderSystem::initialiseContext at /var/lib/jenkins/workspace/ogre-2.2-debbuilder/repo/RenderSystems/GL3Plus/src/OgreGL3PlusRenderSystem.cpp (line 3364)
compose-sim-1        | [ign gazebo-1] [GUI] [Err] [Ogre2RenderEngine.cc:973]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): OpenGL 3.3 is not supported. Please update your graphics card drivers. in GL3PlusRenderSystem::initialiseContext at /var/lib/jenkins/workspace/ogre-2.2-debbuilder/repo/RenderSystems/GL3Plus/src/OgreGL3PlusRenderSystem.cpp (line 3364)
compose-sim-1        | [ign gazebo-1] [GUI] [Err] [Ogre2RenderEngine.cc:973]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): OpenGL 3.3 is not supported. Please update your graphics card drivers. in GL3PlusRenderSystem::initialiseContext at /var/lib/jenkins/workspace/ogre-2.2-debbuilder/repo/RenderSystems/GL3Plus/src/OgreGL3PlusRenderSystem.cpp (line 3364)
compose-sim-1        | [ign gazebo-1] [GUI] [Err] [Ogre2RenderEngine.cc:973]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): OpenGL 3.3 is not supported. Please update your graphics card drivers. in GL3PlusRenderSystem::initialiseContext at /var/lib/jenkins/workspace/ogre-2.2-debbuilder/repo/RenderSystems/GL3Plus/src/OgreGL3PlusRenderSystem.cpp (line 3364)
compose-sim-1        | [ign gazebo-1] [GUI] [Err] [Ogre2RenderEngine.cc:973]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): OpenGL 3.3 is not supported. Please update your graphics card drivers. in GL3PlusRenderSystem::initialiseContext at /var/lib/jenkins/workspace/ogre-2.2-debbuilder/repo/RenderSystems/GL3Plus/src/OgreGL3PlusRenderSystem.cpp (line 3364)
compose-sim-1        | [ign gazebo-1] [GUI] [Err] [Ogre2RenderEngine.cc:973]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): OpenGL 3.3 is not supported. Please update your graphics card drivers. in GL3PlusRenderSystem::initialiseContext at /var/lib/jenkins/workspace/ogre-2.2-debbuilder/repo/RenderSystems/GL3Plus/src/OgreGL3PlusRenderSystem.cpp (line 3364)
compose-sim-1        | [ign gazebo-1] [GUI] [Err] [Ogre2RenderEngine.cc:973]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): OpenGL 3.3 is not supported. Please update your graphics card drivers. in GL3PlusRenderSystem::initialiseContext at /var/lib/jenkins/workspace/ogre-2.2-debbuilder/repo/RenderSystems/GL3Plus/src/OgreGL3PlusRenderSystem.cpp (line 3364)
compose-sim-1        | [ign gazebo-1] [GUI] [Err] [Ogre2RenderEngine.cc:973]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): OpenGL 3.3 is not supported. Please update your graphics card drivers. in GL3PlusRenderSystem::initialiseContext at /var/lib/jenkins/workspace/ogre-2.2-debbuilder/repo/RenderSystems/GL3Plus/src/OgreGL3PlusRenderSystem.cpp (line 3364)
compose-sim-1        | [ign gazebo-1] [GUI] [Err] [Ogre2RenderEngine.cc:973]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): OpenGL 3.3 is not supported. Please update your graphics card drivers. in GL3PlusRenderSystem::initialiseContext at /var/lib/jenkins/workspace/ogre-2.2-debbuilder/repo/RenderSystems/GL3Plus/src/OgreGL3PlusRenderSystem.cpp (line 3364)
compose-sim-1        | [ign gazebo-1] [GUI] [Err] [Ogre2RenderEngine.cc:981] Unable to create the rendering window after [11] attempts.
compose-sim-1        | [ign gazebo-1] [GUI] [Err] [Ogre2RenderEngine.cc:902] Failed to create dummy render window.
compose-sim-1        | [ign gazebo-1] Stack trace (most recent call last) in thread 372:
compose-sim-1        | [ign gazebo-1] #31   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fd7d0eb5784, in QThread::exec()
compose-sim-1        | [ign gazebo-1] #30   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fd7d107d3aa, in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>)
compose-sim-1        | [ign gazebo-1] #29   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fd7d10d6434, in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>)
compose-sim-1        | [ign gazebo-1] #28   Object "/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7fd7ceeb04a2, in g_main_context_iteration
compose-sim-1        | [ign gazebo-1] #27   Object "/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7fd7ceeb03ff, in 
compose-sim-1        | [ign gazebo-1] #26   Object "/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7fd7ceeb017c, in g_main_context_dispatch
compose-sim-1        | [ign gazebo-1] #25   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fd7d10d6e36, in 
[... cut ...] yCategory)0> > > const&, bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
compose-sim-1        | [ign gazebo-1] Segmentation fault (Address not mapped to object [0x220])
compose-sim-1        | [ign gazebo-1] [GUI] [Msg] model:     PMSEscalating to SIGKILL on [Ignition Gazebo Server]
compose-sim-1        | [INFO] [ign gazebo-1]: process has finished cleanly [pid 81]

m00nlab, 2022-08-11:
I found the problem, somehow with --gpus all in docker run, the access to GPU is incomplete. I configured the --runtime nvidia at docker.json and then it started working fine!

closed 2022-08-11 by m00nlab