osrf/mbzirc #186: mbzirc_point_laser measures inf over the sea

fercabe, 2022-07-21:
Hi,

we have mounted a mbzirc_point_laser sensor on the UAV. The sensor is pointing downwards. It works well until the UAV flies over the sea, in that case the measured distance is infinite. 

We discovered that removing the visibility_mask parameters from the sensor definition solve the problem. Can we customize this parameter? Please, consider that we can find in the market laser sensors with multi-echo sensing, suitable for detecting water or glass surfaces, or we can use ultrasonic sensing, as in the DJI M210 or DJI M300 commercial drones.

Thanks
Fernando Caballero

iche033, 2022-07-21:
hi, the visibility mask was added on purpose so that the lidar does not detect water. As the result, the lidar returns inf values because it does not anything within its range.

> Can we customize this parameter? Please, consider that we can find in the market laser sensors with multi-echo sensing, suitable for detecting water or glass surfaces, or we can use ultrasonic sensing, as in the DJI M210 or DJI M300 commercial drones.

This will need to be raised to the organizers. As of now, you won't be able to use the `mbzirc_point_lidar` or customize it for final evaluation. You could try using the depth values from the rgbd camera but it has a much shorter range of 10m.

fercabe, 2022-07-22:
Hi, thanks for clarifying this, Ian.

We already thought about using a RGBD camera to estimate the distance to the seawater, however we discarded this idea for two main reasons: 
- It is a waste of computational resources using such a dense sensor just for point distance computation, in both simulator and team's solution.
- Most RGBD cameras are based on projected IR pattern or IR flashing that do not work on transparent surfaces as water or glass outdoors, so we think it is not much realistic using this sensor for seawater distance computation.

If there is no other choice, of course we will make use of the RGBD sensor for seawater distance estimator. But it sounds like an artificial constraint for us because you can find lightweight laser systems specially designed for distance to water surface estimation in the market, like this one:

https://lasertech.com/themencode-pdf-viewer/?file=https://lasertech.com/wp-content/uploads/FluidLevelMeasurement_TruSense_Brochure.pdf#zoom=auto

fercabe, 2022-07-22:
By the way, when you said "As of now, you won't be able to use the mbzirc_point_lidar or customize it for final evaluation", you mean in general? so we cannot use the point_lidar in our solution?

iche033, 2022-07-25:
> If there is no other choice, of course we will make use of the RGBD sensor for seawater distance estimator. But it sounds like an artificial constraint for us because you can find lightweight laser systems specially designed for distance to water surface estimation in the market, like this one:

I would recommend going with the RGBD camera for seawater distance computation for the final evaluation as a workaround since we won't be able to accommodate  customization at this stage.

> By the way, when you said "As of now, you won't be able to use the mbzirc_point_lidar or customize it for final evaluation", you mean in general? so we cannot use the point_lidar in our solution?

Yes that's correct. The point lidar model was created as a tutorial example to show how customizations can be done, and by default it is not installed. After customization requests were received, a decision was made that Open Robotics would implement some of the most requested features and other cusomizations would no longer be allowed, including the sensor examples of the wiki.

fercabe, 2022-07-27:
Crystal clear, thank you!

closed 2022-07-27 by fercabe