osrf/mbzirc #197: USV visual model does not correspond to physical model

jurajObradovic, 2022-07-28:
It turns out that visual model together with thruster location (usv_base) is rotated 90 degrees in relation to the physical model of the vehicle. Measured top speeds and drag coefficients are far larger in the forward motion then in the side motion.

iche033, 2022-07-28:
the thruster rotation was adjusted in https://github.com/osrf/mbzirc/pull/179. We'll make another release soon. We'll keep the other params unchanged to avoid breaking people's code for now.

mandicLuka, 2022-07-29:
The thruster rotation is fine, but the problem is that USV behaves very strangely. As @jurajObradovic  said, we get top speed sideways that is many times larger then the top speed going forward