osrf/mbzirc #199: Lag in coast environment

shoang22, 2022-07-29:
We're experiencing some time delays in the simulation - our models are lagging when we try to execute any kind of motion (some of us are using the source install if that provides any help).

https://user-images.githubusercontent.com/54875725/181791660-f020e7b0-c88c-48e2-985f-f31d0a9ac288.mp4

ABZ1110, 2022-07-29:
same problem it happens when trying to fly a drone while having a swarm of drones spawned. However, when only working with one drone it works fine.

iche033, 2022-07-29:
What's the CPU / GPU resource usage. Do you have a machine with dedicated GPU? Just to help narrow down the problem, can you try removing the sensor one by one on the UAV and see there is a particular sensor that is causing the issue?