osrf/mbzirc #201: The bitrate of RFComms doesn't support relaying of 1280x960 stream

xirhxq, 2022-08-02:
It is mentioned in [Competition-APIs in wiki](https://github.com/osrf/mbzirc/wiki/Competition-APIs)

> The video stream will be transmitted over the inter-robot communication channel so ensure you are sending images within a reasonable range from the base station located next to the start gate otherwise image data will start dropping. One strategy is to relay the image data to another robot in the field that is closer to the base station and let it be the one sending the live video stream, see [Communications](https://github.com/osrf/mbzirc/wiki/Communications).

Also, it is mentioned in [Communication in wiki](https://github.com/osrf/mbzirc/wiki/Communications)

> There is a maximum data rate allowed among robots communicating over the same network segment (e.g.: 1 Gbps).

So, does that mean that we need to relay our stream by the `/rx` & `/tx` topic on ROS2? The size of a single frame of 1280x960 image is `1280x960x3 = 3686400Bytes = 29491200 bits`, however in the terminal running the environment we can see that

[image]

Also, in `mbzirc/mbzirc_ign/worlds/coast.sdf` I found:

[image]

It seems that it doesn't support such a bit frame of data. So is it a wrong configuration of `coast.sdf`? Or there is some other way to relay stream?

On the other hand, we managed to suppress the image into a smaller size, but seems the `GameLogicPlugin.cc` still gives the target position in the form of `1280x960` pixels. So I wonder what image does `GameLogicPlugin.cc` take to judge?

Best regards.

iche033, 2022-08-02:
When reporting targets, you should specify the image coordinates based on the original resolution of the camera, i.e., 1280x960. Underneath the hood, the GameLogicPlugin will grab an image itself from the HD camera (original uncompressed images) to do the validation.  It is expected that you reduce the size of the image being streamed (if using HD camera) due to comms network limitations, or use VGA camera. For final evaluation, we will be saving some of the images from the image stream that you transmit and visually verify that they are actually valid. Hope that makes sense.

xirhxq, 2022-08-02:
That makes everything clear! Thanks for your effort!