osrf/mbzirc #207: How to start a python script from mbzirc_seed

zorawar12, 2022-08-10:
Hello sir,

The current solution seed which you have provided assumes that the solution plugins are in cpp. However, we have our scripts in python. How are we supposed to modify the launch file "mbzirc_seed.launch.py" in this case?

Thank you

iche033, 2022-08-11:
hi, checking to see if your solution a ROS2 package and the python scripts are ros2 nodes? If so, ros2 launch files support launching nodes written in python as well. Here is a [simple example](https://roboticsbackend.com/ros2-launch-file-example/) I found where you just need to specify your package name and the python script name. Using a ros2 launch file to launch your solution is optional. You can just do `ros2 run <package_name> <python_script>` instead of writing a launch file if that's easier. Alternatively if the entry point of your solution is not a ROS2 package but just a regular python script, you can just do `python3 <path_to_your_script>` in the startup script. Hope that makes sense.

zorawar12, 2022-08-12:
Actually, I am still not sure that, How can I launch robot-specific nodes based on the robot names as you have provided in the seed_solution example directly by using info from the config.yaml.

zorawar12, 2022-08-13:
Due to some error in the beginning of part here, all robots are not getting spawned .....

>Attaching to compose-bridge1-1, compose-bridge10-1, compose-bridge11-1, compose-bridge12-1, compose-bridge13-1, compose-bridge14-1, compose-bridge15-1, compose-bridge16-1, compose-bridge17-1, compose-bridge18-1, compose-bridge2-1, compose-bridge3-1, compose-bridge4-1, compose-bridge5-1, compose-bridge6-1, compose-bridge7-1, compose-bridge8-1, compose-bridge9-1, compose-sim-1, compose-solution1-1, compose-solution10-1, compose-solution11-1, compose-solution12-1, compose-solution13-1, compose-solution14-1, compose-solution15-1, compose-solution16-1, compose-solution17-1, compose-solution18-1, compose-solution2-1, compose-solution3-1, compose-solution4-1, compose-solution5-1, compose-solution6-1, compose-solution7-1, compose-solution8-1, compose-solution9-1
compose-sim-1         | [INFO] [launch]: All log files can be found below /tmp/ign/logs/ros/2022-08-13-17-47-19-205672-969d7f25c6c2-69
compose-sim-1         | [INFO] [launch]: Default logging verbosity is set to INFO
compose-sim-1         | ['erb', 'name=usv', 'slot0=mbzirc_rgbd_camera', 'slot0_pos=0 0 0', 'slot1=mbzirc_naive_3d_scanning_radar', 'slot1_pos=0 0 0', 'wavefieldSize=0', 'arm=mbzirc_oberon7_arm', 'arm_slot=0', '/home/developer/mbzirc_ws/install/share/mbzirc_ign/models/usv/model.sdf.erb']
compose-sim-1         | ['erb', 'name=fixedWings_1', 'slot0=mbzirc_rgbd_camera', 'slot0_pos=0 0 0', 'slot3=mbzirc_rgbd_camera', 'slot3_pos=0 0 0', 'capacity=0.34662045060658575', '/home/developer/mbzirc_ws/install/share/mbzirc_ign/models/mbzirc_fixed_wing/model.sdf.erb']
compose-sim-1         | ['erb', 'name=fixedWings_2', 'slot0=mbzirc_rgbd_camera', 'slot0_pos=0 0 0', 'slot3=mbzirc_rgbd_camera', 'slot3_pos=0 0 0', 'capacity=0.34662045060658575', '/home/developer/mbzirc_ws/install/share/mbzirc_ign/models/mbzirc_fixed_wing/model.sdf.erb']
compose-sim-1         | ['erb', 'name=fixedWings_3', 'slot0=mbzirc_rgbd_camera', 'slot0_pos=0 0 0', 'slot3=mbzirc_rgbd_camera', 'slot3_pos=0 0 0', 'capacity=0.34662045060658575', '/home/developer/mbzirc_ws/install/share/mbzirc_ign/models/mbzirc_fixed_wing/model.sdf.erb']
compose-sim-1         | ['erb', 'name=fixedWings_4', 'slot0=mbzirc_rgbd_camera', 'slot0_pos=0 0 0', 'slot3=mbzirc_rgbd_camera', 'slot3_pos=0 0 0', 'capacity=0.34662045060658575', '/home/developer/mbzirc_ws/install/share/mbzirc_ign/models/mbzirc_fixed_wing/model.sdf.erb']
compose-sim-1         | ['erb', 'name=hexarotor_1', 'slot0=mbzirc_rgbd_camera', 'slot0_pos=0 0 0', 'slot6=mbzirc_rgbd_camera', 'slot6_pos=0 0 0', 'slot7=mbzirc_3d_lidar', 'slot7_pos=0 0 0', 'capacity=0.17331022530329288', 'gripper=mbzirc_oberon7_gripper_hexarotor_1', '/home/developer/mbzirc_ws/install/share/mbzirc_ign/models/mbzirc_hexrotor/model.sdf.erb']
compose-sim-1         | ['erb', 'name=hexarotor_2', 'slot0=mbzirc_rgbd_camera', 'slot0_pos=0 0 0', 'slot6=mbzirc_rgbd_camera', 'slot6_pos=0 0 0', 'slot7=mbzirc_3d_lidar', 'slot7_pos=0 0 0', 'capacity=0.17331022530329288', 'gripper=mbzirc_oberon7_gripper_hexarotor_2', '/home/developer/mbzirc_ws/install/share/mbzirc_ign/models/mbzirc_hexrotor/model.sdf.erb']
compose-sim-1         | ['erb', 'name=hexarotor_3', 'slot0=mbzirc_rgbd_camera', 'slot0_pos=0 0 0', 'slot6=mbzirc_rgbd_camera', 'slot6_pos=0 0 0', 'slot7=mbzirc_3d_lidar', 'slot7_pos=0 0 0', 'capacity=0.17331022530329288', 'gripper=mbzirc_oberon7_gripper_hexarotor_3', '/home/developer/mbzirc_ws/install/share/mbzirc_ign/models/mbzirc_hexrotor/model.sdf.erb']
compose-sim-1         | ['erb', 'name=quadrotor_1', 'slot0=mbzirc_rgbd_camera', 'slot0_pos=0 0 0', 'slot3=mbzirc_3d_lidar', 'slot3_pos=0 0 0', 'capacity=0.17331022530329288', '/home/developer/mbzirc_ws/install/share/mbzirc_ign/models/mbzirc_quadrotor/model.sdf.erb']
compose-sim-1         | ['erb', 'name=quadrotor_2', 'slot0=mbzirc_rgbd_camera', 'slot0_pos=0 0 0', 'slot3=mbzirc_3d_lidar', 'slot3_pos=0 0 0', 'capacity=0.17331022530329288', '/home/developer/mbzirc_ws/install/share/mbzirc_ign/models/mbzirc_quadrotor/model.sdf.erb']
compose-sim-1         | ['erb', 'name=quadrotor_3', 'slot0=mbzirc_rgbd_camera', 'slot0_pos=0 0 0', 'slot3=mbzirc_3d_lidar', 'slot3_pos=0 0 0', 'capacity=0.17331022530329288', '/home/developer/mbzirc_ws/install/share/mbzirc_ign/models/mbzirc_quadrotor/model.sdf.erb']
compose-sim-1         | ['erb', 'name=quadrotor_4', 'slot0=mbzirc_rgbd_camera', 'slot0_pos=0 0 0', 'slot3=mbzirc_3d_lidar', 'slot3_pos=0 0 0', 'capacity=0.17331022530329288', '/home/developer/mbzirc_ws/install/share/mbzirc_ign/models/mbzirc_quadrotor/model.sdf.erb']
compose-sim-1         | ['erb', 'name=quadrotor_5', 'slot0=mbzirc_rgbd_camera', 'slot0_pos=0 0 0', 'slot3=mbzirc_3d_lidar', 'slot3_pos=0 0 0', 'capacity=0.17331022530329288', '/home/developer/mbzirc_ws/install/share/mbzirc_ign/models/mbzirc_quadrotor/model.sdf.erb']
compose-sim-1         | ['erb', 'name=quadrotor_6', 'slot0=mbzirc_rgbd_camera', 'slot0_pos=0 0 0', 'slot6=mbzirc_rgbd_camera', 'slot6_pos=0 0 0', 'capacity=0.17331022530329288', '/home/developer/mbzirc_ws/install/share/mbzirc_ign/models/mbzirc_quadrotor/model.sdf.erb']
compose-sim-1         | ['erb', 'name=quadrotor_7', 'slot0=mbzirc_rgbd_camera', 'slot0_pos=0 0 0', 'slot6=mbzirc_rgbd_camera', 'slot6_pos=0 0 0', 'capacity=0.17331022530329288', '/home/developer/mbzirc_ws/install/share/mbzirc_ign/models/mbzirc_quadrotor/model.sdf.erb']
[... cut ...]
compose-sim-1         |       <robotNamespace>model/quadrotor_9</robotNamespace>
compose-sim-1         |       <commandSubTopic>cmd_vel</commandSubTopic>
compose-sim-1         |       <motorControlPubTopic>command/motor_speed</motorControlPubTopic>
compose-sim-1         |       <enableSubTopic>velocity_controller/enable</enableSubTopic>
compose-sim-1         |       <comLinkName>base_link</comLinkName>
compose-sim-1         |       <velocityGain>2.7 2.7 2.7</velocityGain>
compose-sim-1         |       <attitudeGain>2 3 0.15</attitudeGain>

zorawar12, 2022-08-13:
>compose-sim-1         |       <angularRateGain>0.4 0.52 0.18</angularRateGain>
compose-sim-1         |       <maximumLinearAcceleration>1 1 2</maximumLinearAcceleration>
compose-sim-1         |       <maximumLinearVelocity>5 5 5</maximumLinearVelocity>
compose-sim-1         |       <maximumAngularVelocity>3 3 3</maximumAngularVelocity>
compose-sim-1         |       <linearVelocityNoiseMean>0 0 0.05</linearVelocityNoiseMean>
compose-sim-1         |       
compose-sim-1         |       
compose-sim-1         |       
compose-sim-1         |       <linearVelocityNoiseStdDev>0.1105 0.1261 0.00947</linearVelocityNoiseStdDev>
compose-sim-1         |       <angularVelocityNoiseMean>0 0 0</angularVelocityNoiseMean>
compose-sim-1         |       
compose-sim-1         |       
compose-sim-1         |       <angularVelocityNoiseStdDev>0.004 0.004 0.004</angularVelocityNoiseStdDev>
compose-sim-1         |       <rotorConfiguration>
compose-sim-1         |         <rotor>
compose-sim-1         |           <jointName>rotor_0_joint</jointName>
compose-sim-1         |           <forceConstant>8.54858e-06</forceConstant>
compose-sim-1         |           <momentConstant>0.016</momentConstant>
compose-sim-1         |           <direction>1</direction>
compose-sim-1         |         </rotor>
compose-sim-1         |         <rotor>
compose-sim-1         |           <jointName>rotor_1_joint</jointName>
compose-sim-1         |           <forceConstant>8.54858e-06</forceConstant>
compose-sim-1         |           <momentConstant>0.016</momentConstant>
compose-sim-1         |           <direction>1</direction>
compose-sim-1         |         </rotor>
compose-sim-1         |         <rotor>
compose-sim-1         |           <jointName>rotor_2_joint</jointName>
compose-sim-1         |           <forceConstant>8.54858e-06</forceConstant>
compose-sim-1         |           <momentConstant>0.016</momentConstant>
compose-sim-1         |           <direction>-1</direction>
compose-sim-1         |         </rotor>
compose-sim-1         |         <rotor>
compose-sim-1         |           <jointName>rotor_3_joint</jointName>
compose-sim-1         |           <forceConstant>8.54858e-06</forceConstant>
compose-sim-1         |           <momentConstant>0.016</momentConstant>
compose-sim-1         |           <direction>-1</direction>
compose-sim-1         |         </rotor>
compose-sim-1         |       </rotorConfiguration>
compose-sim-1         |     </plugin>
compose-sim-1         |     
compose-sim-1         |     
compose-sim-1         |     <plugin filename="libignition-gazebo-linearbatteryplugin-system.so"
compose-sim-1         |       name="ignition::gazebo::systems::LinearBatteryPlugin">
compose-sim-1         |       <battery_name>linear_battery</battery_name>
compose-sim-1         |       <voltage>12.694</voltage>
compose-sim-1         |       <open_circuit_voltage_constant_coef>12.694</open_circuit_voltage_constant_coef>
compose-sim-1         |       <open_circuit_voltage_linear_coef>0</open_circuit_voltage_linear_coef>
compose-sim-1         |       <initial_charge>0.17331022530329288</initial_charge>
compose-sim-1         |       <capacity>0.17331022530329288</capacity>
compose-sim-1         |       <resistance>0.061523</resistance>
compose-sim-1         |       <smooth_current_tau>1.9499</smooth_current_tau>
compose-sim-1         |       <power_load>6.6</power_load>
compose-sim-1         |       <start_on_motion>true</start_on_motion>
compose-sim-1         |       <fix_issue_225>true</fix_issue_225>
compose-sim-1         |     </plugin>
compose-sim-1         | 
compose-sim-1         |     
compose-sim-1         |     <plugin
compose-sim-1         |       filename="ignition-gazebo-comms-endpoint-system"
compose-sim-1         |       name="ignition::gazebo::systems::CommsEndpoint">
compose-sim-1         |       <address>quadrotor_9</address>
compose-sim-1         |       <topic>model/quadrotor_9/rx</topic>
compose-sim-1         |     </plugin>
compose-sim-1         |  </model>
compose-sim-1         | </sdf>
compose-sim-1         |  -name quadrotor_9 -allow_renaming false -x -1510.57 -y -1.76 -z 4.3 -R 0 -P 0 -Y 0 --ros-args'].
compose-sim-1         | [ign gazebo-1] [Msg] Ignition Gazebo Server v6.10.0
compose-sim-1         | [ign gazebo-1] [Msg] Loading SDF world file[/home/developer/mbzirc_ws/install/share/mbzirc_ign/worlds/coast.sdf].
compose-sim-1         | [ign gazebo-1] [Msg] Serving entity system service on [/entity/system/add]
[... 1403 lines cut ...]
compose-sim-1         | [ign gazebo-1] Msg [NameManager::issueNewName] (Skeleton::BodyNode | large_crate_d_duplicate) The name [caster_back_left] is a duplicate, so it has been renamed to [caster_back_left(1)]
compose-sim-1         | [ign gazebo-1] Msg [NameManager::issueNewName] (Skeleton::Joint | large_crate_d_duplicate) The name [joint_caster_back_left] is a duplicate, so it has been renamed to [joint_caster_back_left(1)]
compose-sim-1         | [ign gazebo-1] Msg [NameManager::issueNewName] (Skeleton::N4dart8dynamics9ShapeNodeE | large_crate_d_duplicate) The name [caster_back_left:caster_back_left_col] is a duplicate, so it has been renamed to [caster_back_left:caster_back_left_col(1)]
compose-sim-1         | [ign gazebo-1] Msg [NameManager::issueNewName] (Skeleton::BodyNode | large_crate_d_duplicate) The name [suspension_back_right] is a duplicate, so it has been renamed to [suspension_back_right(1)]
[... cut ...]
compose-sim-1         | [ign gazebo-1] Msg [NameManager::issueNewName] (Skeleton::N4dart8dynamics9ShapeNodeE | large_crate_d_duplicate) The name [link:Box016(1)(1)] is a duplicate, so it has been renamed to [link:Box016(1)(1)(1)]
compose-sim-1         | [ign gazebo-1] Msg [NameManager::issueNewName] (Skeleton::N4dart8dynamics9ShapeNodeE | large_crate_d_duplicate) The name [link:Box015(1)(1)] is a duplicate, so it has been renamed to [link:Box015(1)(1)(1)]
compose-sim-1         | [ign gazebo-1] Msg [NameManager::issueNewName] (Skeleton::N

zorawar12, 2022-08-13:
>compose-sim-1         | [ign gazebo-1] Msg [NameManager::issueNewName] (Skeleton::Joint | large_ammo_can_f_duplicate) The name [joint_caster_front_right] is a duplicate, so it has been renamed to [joint_caster_front_right(1)]
compose-sim-1         | [ign gazebo-1] Msg [NameManager::issueNewName] (Skeleton::N4dart8dynamics9ShapeNodeE | large_ammo_can_f_duplicate) The name [caster_front_right:caster_front_right_col] is a duplicate, so it has been renamed to [caster_front_right:caster_front_right_col(1)]
compose-sim-1         | [ign gazebo-1] Msg [NameManager::issueNewName] (Skeleton::BodyNode | large_ammo_can_f_duplicate) The name [suspension_back_left] is a duplicate, so it has been renamed to [suspension_back_left(1)]
compose-sim-1         | [ign gazebo-1] Msg [NameManager::issueNewName] (Skeleton::Joint | large_ammo_can_f_duplicate) The name [joint_suspension_back_left] is a duplicate, so it has been renamed to [joint_suspension_back_left(1)]
compose-sim-1         | [ign gazebo-1] Msg [NameManager::issueNewName] (Skeleton::BodyNode | large_ammo_can_f_duplicate) The name [caster_back_left] is a duplicate, so it has been renamed to [caster_back_left(1)]
compose-sim-1         | [ign gazebo-1] Msg [NameManager::issueNewName] (Skeleton::Joint | large_ammo_can_f_duplicate) The name [joint_caster_back_left] is a duplicate, so it has been renamed to [joint_caster_back_left(1)]
compose-sim-1         | [ign gazebo-1] Msg [NameManager::issueNewName] (Skeleton::N4dart8dynamics9ShapeNodeE | large_ammo_can_f_duplicate) The name [caster_back_left:caster_back_left_col] is a duplicate, so it has been renamed to [caster_back_left:caster_back_left_col(1)]
compose-sim-1         | [ign gazebo-1] Msg [NameManager::issueNewName] (Skeleton::BodyNode | large_ammo_can_f_duplicate) The name [suspension_back_right] is a duplicate, so it has been renamed to [suspension_back_right(1)]
compose-sim-1         | [ign gazebo-1] Msg [NameManager::issueNewName] (Skeleton::Joint | large_ammo_can_f_duplicate) The name [joint_suspension_back_right] is a duplicate, so it has been renamed to [joint_suspension_back_right(1)]
compose-sim-1         | [ign gazebo-1] Msg [NameManager::issueNewName] (Skeleton::BodyNode | large_ammo_can_f_duplicate) The name [caster_back_right] is a duplicate, so it has been renamed to [caster_back_right(1)]
compose-sim-1         | [ign gazebo-1] Msg [NameManager::issueNewName] (Skeleton::Joint | large_ammo_can_f_duplicate) The name [joint_caster_back_right] is a duplicate, so it has been renamed to [joint_caster_back_right(1)]
compose-sim-1         | [ign gazebo-1] Msg [NameManager::issueNewName] (Skeleton::N4dart8dynamics9ShapeNodeE | large_ammo_can_f_duplicate) The name [caster_back_right:caster_back_right_col] is a duplicate, so it has been renamed to [caster_back_right:caster_back_right_col(1)]
compose-sim-1         | [ign gazebo-1] Msg [NameManager::issueNewName] (Skeleton::Joint | large_ammo_can_f_duplicate) The name [fixed] is a duplicate, so it has been renamed to [fixed(1)]
compose-sim-1         | [ign gazebo-1] Msg [NameManager::issueNewName] (Skeleton::Joint | large_ammo_can_f_duplicate) The name [fixed(1)] is a duplicate, so it has been renamed to [fixed(1)(1)]
compose-sim-1         | [ign gazebo-1] Msg [NameManager::issueNewName] (Skeleton::Joint | large_ammo_can_f_duplicate) The name [fixed(1)(1)] is a duplicate, so it has been renamed to [fixed(1)(1)(1)]
compose-sim-1         | [ign gazebo-1] [Dbg] [Physics.cc:1593] Creating detachable joint [1506]
compose-sim-1         | [ign gazebo-1] [Dbg] [Physics.cc:1593] Creating detachable joint [1507]
compose-sim-1         | [ign gazebo-1] Msg [NameManager::issueNewName] (Skeleton::N4dart8dynamics9ShapeNodeE | small_dry_bag_handle_g_duplicate) The name [link:Cylinder001] is a duplicate, so it has been renamed to [link:Cylinder001(1)]
compose-sim-1         | [ign gazebo-1] Msg [NameManager::issueNewName] (Skeleton::N4dart8dynamics9ShapeNodeE | small_dry_bag_handle_g_duplicate) The name [link:Cylinder003] is a duplicate, so it has been renamed to [link:Cylinder003(1)]
compose-sim-1         | [ign gazebo-1] Msg [NameManager::issueNewName] (Skeleton::N4dart8dynamics9ShapeNodeE | small_dry_bag_handle_g_duplicate) The name [link:Cylinder004] is a duplicate, so it has been renamed to [link:Cylinder004(1)]
compose-sim-1         | [ign gazebo-1] Msg [NameManager::issueNewName] (Skeleton::N4dart8dynamics9ShapeNodeE | small_dry_bag_handle_g_duplicate) The name [link:Box001] is a duplicate, so it has been renamed to [link:Box001(1)]
compose-sim-1         | [ign gazebo-1] Msg [NameManager::issueNewName] (Skeleton::N4dart8dynamics9ShapeNodeE | small_dry_bag_handle_g_duplicate) The name [link:Box002] is a duplicate, so it has been renamed to [link:Box002(1)]
compose-sim-1         | [ign gazebo-1] Msg [NameManager::issueNewName] (Skeleton::N4dart8dynamics9ShapeNodeE | small_dry_bag_handle_g_duplicate) The name [link:Box003] is a duplicate, so it has been renamed to [link:Box003(1)]
compose-sim-1         | [ign gazebo-1] Msg [NameManager::issueNewName] (Skeleton::N4dart8dynamics9ShapeNodeE | small_dry_bag_handle_g_duplicate) The name [link:Box004] is a duplicate, so it has been renamed to [link:Box004(1)]
compose-sim-1         | [ign gazebo-1] Msg [NameManager::issueNewName] (Skeleton::N4dart8dynamics9ShapeNodeE | small_dry_bag_handle_g_duplicate) The name [link:Box005] is a duplicate, so it has been renamed to [link:Box005(1)]
[... cut ...]
compose-bridge9-1     | [parameter_bridge-3] [INFO] [1660393073.973023841] [quadrotor_1.ros_ign_bridge]: Creating IGN->ROS Bridge: [/world/coast/model/quadrotor_1/model/sensor_0/link/sensor_link/sensor/camera/camera_info (ignition.msgs.CameraInfo) -> /world/coast/model/quadrotor_1/model/sensor_0/link/sensor_link/sensor/camera/camera_info (sensor_msgs/msg/CameraInfo)] (Lazy 0)
compose-bridge9-1     | [parameter_bridge-3] [INFO] [1660393073.973991240] [quadrotor_1.ros_ign_bridge]: Creating IGN->ROS Bridge: [/world/coast/model/quadrotor_1/model/sensor_0/link/sensor_link/sensor/camera/depth_image (ignition.msgs.

iche033, 2022-08-16:
> Actually, I am still not sure that, How can I launch robot-specific nodes based on the robot names as you have provided in the seed_solution example directly by using info from the config.yaml.

Not sure if this helps but here is what we are doing:

1. Parse the submitted `config.yaml` to get all the robot names. 
1. Loop through each robot, launch a docker container using your submitted docker solution and pass the robot name to your startup script
1. Your start up script should have some sort of logic that does:

    ```
    # pseudo code
    robot_name = arg
    if  robot_name == "my_usv"
      launch usv node
    else if robot_name == "my_quad_1"
      launch quad_1 node
    ```

In our example seed solution, we did not put these `if` checks in the startup script. Instead we pass the robot name arg to the launch file and check what node to bring up using a [filter](https://github.com/osrf/mbzirc/blob/main/mbzirc_seed/launch/mbzirc_seed.launch.py#L75) inside the launch file.

> Also, if I directly launch Nodes/include other launch files in the mbzirc_seed.launch.py instead of launching them with composableNode format, won't it be a problem for you in terms of making sure that the Nodes aren't accessing data from other robots bypassing the simulation communication structure (I mean the TX/RX topics)?

In terms of making sure nodes aren't accesing data from other robots, this will not be a problem for us. In final evaluation, we have set up networking so that the solution container launched for a robot is only able to see its own topics and not able to talk to another container launched for a different robot.

> Due to some error in the beginning of part here, all robots are not getting spawned .....

From the console output, looks like your code is expecting a different type here?

```
compose-solution1-1 | [usv_g2g.py-1] File "/home/developer/mbzirc_ws/install/lib/mbzirc_seed/usv_g2g.py", line 88, in task
compose-solution1-1 | [usv_g2g.py-1] self.poses.append([self.x,self.y,self.z])
compose-solution1-1 | [usv_g2g.py-1] AttributeError: 'USV' object has no attribute 'x'
```

zorawar12, 2022-08-17:
Hello sir,

>From the console output, looks like your code is expecting a different type here?

That part I understand but before that some robots are not getting spawned at all. If I am spawning 20 robots in total, I can see only 8 in the simulator.

What could be the reason?

iche033, 2022-08-18:
> That part I understand but before that some robots are not getting spawned at all. If I am spawning 20 robots in total, I can see only 8 in the simulator.

You could be hitting a ROS2 DDS issue? Do you see any `"Failed to find a free participant index for domain"` error msgs printed in the console?

iche033, 2022-08-18:
if you're using the docker compose file, try removing the `ROS_LOCAL_HOST_ONLY` [line](https://github.com/osrf/mbzirc/blob/v4/docker/compose/gen_docker_compose.py#L79)

zorawar12, 2022-08-18:
> You could be hitting a ROS2 DDS issue? Do you see any `"Failed to find a free participant index for domain"` error msgs printed in the console?

Yes I see this error when using the compose file, also without the ROS_LOCAL_HOST_ONLY line, I am getting display errors (OGRE issue):

>compose-sim-1         | [ign gazebo-1] [GUI] [Err] [Ogre2RenderEngine.cc:973]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): OpenGL 3.3 is not supported. Please update your graphics card drivers. in GL3PlusRenderSystem::initialiseContext at /var/lib/jenkins/workspace/ogre-2.2-debbuilder/repo/RenderSystems/GL3Plus/src/OgreGL3PlusRenderSystem.cpp (line 3364)
compose-sim-1         | [ign gazebo-1] [GUI] [Err] [Ogre2RenderEngine.cc:973]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): OpenGL 3.3 is not supported. Please update your graphics card drivers. in GL3PlusRenderSystem::initialiseContext at /var/lib/jenkins/workspace/ogre-2.2-debbuilder/repo/RenderSystems/GL3Plus/src/OgreGL3PlusRenderSystem.cpp (line 3364)
compose-sim-1         | [ign gazebo-1] [GUI] [Err] [Ogre2RenderEngine.cc:973]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): OpenGL 3.3 is not supported. Please update your graphics card drivers. in GL3PlusRenderSystem::initialiseContext at /var/lib/jenkins/workspace/ogre-2.2-debbuilder/repo/RenderSystems/GL3Plus/src/OgreGL3PlusRenderSystem.cpp (line 3364)
compose-sim-1         | [ign gazebo-1] [GUI] [Err] [Ogre2RenderEngine.cc:973]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): OpenGL 3.3 is not supported. Please update your graphics card drivers. in GL3PlusRenderSystem::initialiseContext at /var/lib/jenkins/workspace/ogre-2.2-debbuilder/repo/RenderSystems/GL3Plus/src/OgreGL3PlusRenderSystem.cpp (line 3364)
compose-sim-1         | [ign gazebo-1] [GUI] [Err] [Ogre2RenderEngine.cc:973]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): OpenGL 3.3 is not supported. Please update your graphics card drivers. in GL3PlusRenderSystem::initialiseContext at /var/lib/jenkins/workspace/ogre-2.2-debbuilder/repo/RenderSystems/GL3Plus/src/OgreGL3PlusRenderSystem.cpp (line 3364)
compose-sim-1         | [ign gazebo-1] [GUI] [Err] [Ogre2RenderEngine.cc:973]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): OpenGL 3.3 is not supported. Please update your graphics card drivers. in GL3PlusRenderSystem::initialiseContext at /var/lib/jenkins/workspace/ogre-2.2-debbuilder/repo/RenderSystems/GL3Plus/src/OgreGL3PlusRenderSystem.cpp (line 3364)
compose-sim-1         | [ign gazebo-1] [GUI] [Err] [Ogre2RenderEngine.cc:973]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): OpenGL 3.3 is not supported. Please update your graphics card drivers. in GL3PlusRenderSystem::initialiseContext at /var/lib/jenkins/workspace/ogre-2.2-debbuilder/repo/RenderSystems/GL3Plus/src/OgreGL3PlusRenderSystem.cpp (line 3364)
compose-sim-1         | [ign gazebo-1] [GUI] [Err] [Ogre2RenderEngine.cc:973]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): OpenGL 3.3 is not supported. Please update your graphics card drivers. in GL3PlusRenderSystem::initialiseContext at /var/lib/jenkins/workspace/ogre-2.2-debbuilder/repo/RenderSystems/GL3Plus/src/OgreGL3PlusRenderSystem.cpp (line 3364)
compose-sim-1         | [ign gazebo-1] [GUI] [Err] [Ogre2RenderEngine.cc:973]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): OpenGL 3.3 is not supported. Please update your graphics card drivers. in GL3PlusRenderSystem::initialiseContext at /var/lib/jenkins/workspace/ogre-2.2-debbuilder/repo/RenderSystems/GL3Plus/src/OgreGL3PlusRenderSystem.cpp (line 3364)
compose-sim-1         | [ign gazebo-1] [GUI] [Err] [Ogre2RenderEngine.cc:973]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): OpenGL 3.3 is not supported. Please update your graphics card drivers. in GL3PlusRenderSystem::initialiseContext at /var/lib/jenkins/workspace/ogre-2.2-debbuilder/repo/RenderSystems/GL3Plus/src/OgreGL3PlusRenderSystem.cpp (line 3364)
compose-sim-1         | [ign gazebo-1] [GUI] [Err] [Ogre2RenderEngine.cc:981] Unable to create the rendering window after [11] attempts.
compose-sim-1         | [ign gazebo-1] [GUI] [Err] [Ogre2RenderEngine.cc:902] Failed to create dummy render window.

iche033, 2022-08-19:
that looks like the graphics driver issue in docker container. One thing to try is add the env var `MESA_GL_VERSION_OVERRIDE=3.3` and see if you can get it to start. It's likely going to be doing software rendering.

zorawar12, 2022-08-21:
> that looks like the graphics driver issue in docker container. One thing to try is add the env var `MESA_GL_VERSION_OVERRIDE=3.3` and see if you can get it to start. It's likely going to be doing software rendering.

I am getting the following error now:

>INFO] [1661121470.964344392] [ros_ign_gazebo]: Requesting list of world names.
compose-sim-1         | [ign gazebo-1] qt.qpa.xcb: could not connect to display :1
compose-sim-1         | [ign gazebo-1] qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
compose-sim-1         | [ign gazebo-1] This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.
compose-sim-1         | [ign gazebo-1] 
compose-sim-1         | [ign gazebo-1] Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, xcb.
compose-sim-1         | [ign gazebo-1] 
compose-sim-1         | [ign gazebo-1] Stack trace (most recent call last):
compose-sim-1         | [ign gazebo-1] #31   Object "/usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f69f1bb745d, in ruby_run_node
compose-sim-1         | [ign gazebo-1] #30   Object "/usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f69f1bb25f0, in 
compose-sim-1         | [ign gazebo-1] #29   Object "/usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f69f1d21cf1, in rb_vm_exec
compose-sim-1         | [ign gazebo-1] #28   Object "/usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f69f1d1b5e0, in 
compose-sim-1         | [ign gazebo-1] #27   Object "/usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f69f1d0a805, in 
compose-sim-1         | [ign gazebo-1] #26   Object "/usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f69f1d242da, in 
compose-sim-1         | [ign gazebo-1] #25   Object "/usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f69f1d174b4, in 
compose-sim-1         | [ign gazebo-1] #24   Object "/usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f69f1c753be, in 
compose-sim-1         | [ign gazebo-1] #23   Object "/usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f69f1bb5742, in rb_protect
compose-sim-1         | [ign gazebo-1] #22   Object "/usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f69f1d2ef80, in rb_yield
compose-sim-1         | [ign gazebo-1] #21   Object "/usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f69f1d21cf1, in rb_vm_exec
compose-sim-1         | [ign gazebo-1] #20   Object "/usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f69f1d1b5e0, in 
compose-sim-1         | [ign gazebo-1] #19   Object "/usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f69f1d0a805, in 
compose-sim-1         | [ign gazebo-1] #18   Object "/usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f69f1d242da, in 
compose-sim-1         | [ign gazebo-1] #17   Object "/usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f69f1d174b4, in 
compose-sim-1         | [ign gazebo-1] #16   Object "/usr/lib/x86_64-linux-gnu/ruby/2.7.0/fiddle.so", at 0x7f69ee0f7714, in 
compose-sim-1         | [ign gazebo-1] #15   Object "/usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f69f1ce8929, in rb_nogvl
compose-sim-1         | [ign gazebo-1] #14   Object "/usr/lib/x86_64-linux-gnu/ruby/2.7.0/fiddle.so", at 0x7f69ee0f78fb, in 
compose-sim-1         | [ign gazebo-1] #13   Object "/usr/lib/x86_64-linux-gnu/libffi.so.7", at 0x7f69ee0ec409, in 
compose-sim-1         | [ign gazebo-1] #12   Object "/usr/lib/x86_64-linux-gnu/libffi.so.7", at 0x7f69ee0ecff4, in 
compose-sim-1         | [ign gazebo-1] #11   Object "/usr/lib/x86_64-linux-gnu/libignition-gazebo6-ign.so.6.10.0", at 0x7f69ed6857a0, in runGui
compose-sim-1         | [ign gazebo-1] #10   Object "/usr/lib/x86_64-linux-gnu/libignition-gazebo6-gui.so.6", at 0x7f69ed4884ff, in ignition::gazebo::v6::gui::runGui(int&, char**, char const*, char const*)
compose-sim-1         | [ign gazebo-1] #9    Object "/usr/lib/x86_64-linux-gnu/libignition-gazebo6-gui.so.6", at 0x7f69ed4866de, in ignition::gazebo::v6::gui::createGui(int&, char**, char const*, char const*, bool, char const*)
compose-sim-1         | [ign gazebo-1] #8    Object "/usr/lib/x86_64-linux-gnu/libignition-gui6.so.6", at 0x7f69ebedcf57, in ignition::gui::Application::Application(int&, char**, ignition::gui::WindowType)
compose-sim-1         | [ign gazebo-1] #7    Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f69eb97a3bc, in QApplicationPrivate::init()
compose-sim-1         | [ign gazebo-1] #6    Object "/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5", at 0x7f69ea987542, in QGuiApplicationPrivate::init()
compose-sim-1         | [ign gazebo-1] #5    Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f69ec1d6f54, in QCoreApplicationPrivate::init()
compose-sim-1         | [ign gazebo-1] #4    Object "/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5", at 0x7f69ea985707, in QGuiApplicationPrivate::createEventDispatcher()
compose-sim-1         | [ign gazebo-1] #3    Object "/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5", at 0x7f69ea9847ad, in QGuiApplicationPrivate::createPlatformIntegration()
compose-sim-1         | [ign gazebo-1] #2    Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f69ebfcfaac, in QMessageLogger::fatal(char const*, ...) const
compose-sim-1         | [ign gazebo-1] #1    Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7f69f1939858, in abort
compose-sim-1         | [ign gazebo-1] #0    Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7f69f195a00b, in gsignal
[... cut ...]
compose-solution5-1 exited with code 0
compose-solution18-1 exited with code 0
compose-solution9-1 exited with code 0
compose-solution17-1 exited with code 0
compose-solution14-1 exited with code 0
compose-solution7-1 exited with code 0
compose-solution16-1 exited with code 0
compose-solution4-1 exited with code 0
compose-solution8-1 exited with code 0
compose-solution13-1 exited with code 0
compose-solution15-1 exited with code 0
compose-solution2-1 exited with code 0
compose-solution12-1 exited with code 0

iche033, 2022-08-23:
One workaround for this is to run the docker compose simulation headless and attach a gui to it from another container, e.g.

```
python3 gen_docker_compose.py --config <your_config> --image <your_image> --headless 1
```

Run docker compose with the new yaml file:

```
docker compose -f mbzirc_compose.yaml up
```

Then launch a docker container:

```
bash run.bash osrf/mbzirc:mbzirc_sim_latest
```

Attach a gui to it:

```
IGN_PARTITION=sim ign gazebo -v 4 -g
```

The downside is that the simulator maynot be able to find certain meshes (gripper) in the GUI but they should be there in the simulation. It's more of a GUI visualization issue.