osrf/mbzirc #208: Access /mbzirc/target/stream/status topic

fbenti, 2022-08-12:
In the Competition APIs section, it is mentioned that " During final evaluation, each robot  will only be able to access stream topics in its namespace." This means that we cannot access the /mbzirc/target/stream/status topic from any drone to receive feedback and use them to plan the operations? Or is it the only topic to which we can always subscribe to?

fbenti, 2022-08-12:
Or is that topic intended to be accessible only by drones that are within communication range with the base station?

iche033, 2022-08-15:
the robots will only be able to access its own namespaced stream topics:

```
/<robot_name>/mbzirc/target/stream/start
/<robot_name>/mbzirc/target/stream/report
```

All robots should be able to subscribe to `/mbzirc/target/stream/status` without being affected by comms range

closed 2022-08-16 by fbenti