osrf/mbzirc #209: Simulator shuts down when reporting vessel

chalseokim, 2022-08-13:
Hello, we are currently attempting to get confirmation from the base station. 
We are having difficulty with this as the simulator shuts down when we use the commandline from the wiki page which are
`ros2 run image_transport republish raw --ros-args --remap in:=/quadrotor_17/slot3/optical/image_raw --remap out:=/quadrotor_17/mbzirc/target/stream/start
`
`ros2 topic pub --once /quadrotor_1/mbzirc/target/stream/report ros_ign_interfaces/msg/StringVec '{data: ["vessel", "0", "0"]}'
`

Below is the terminal output right after the report, the second command.

Thank you in advance. 

```
[ign gazebo-1] Stack trace (most recent call last) in thread 619363:
[ign gazebo-1] #15   Object "[0xffffffffffffffff]", at 0xffffffffffffffff, in 
[ign gazebo-1] #14   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f4f59e24132, in clone
[ign gazebo-1] #13   Object "/lib/x86_64-linux-gnu/libpthread.so.0", at 0x7f4f59cea608, in 
[ign gazebo-1] #12   Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f4f55cf0de3, in 
[ign gazebo-1] #11   Object "/usr/lib/x86_64-linux-gnu/ign-gazebo-6/plugins/libignition-gazebo-sensors-system.so", at 0x7f4f1ffffda7, in ignition::gazebo::v6::systems::SensorsPrivate::RenderThread()
[ign gazebo-1] #10   Object "/usr/lib/x86_64-linux-gnu/ign-gazebo-6/plugins/libignition-gazebo-sensors-system.so", at 0x7f4f1ffff710, in ignition::gazebo::v6::systems::SensorsPrivate::RunOnce()
[ign gazebo-1] #9    Object "/home/user/mbzirc_ws/install/lib/libGameLogicPlugin.so", at 0x7f4f1db1d094, in std::_Function_handler<void (), std::_Bind<void (mbzirc::GameLogicPluginPrivate::*(mbzirc::GameLogicPluginPrivate*))()> >::_M_invoke(std::_Any_data const&)
[ign gazebo-1] #8    Object "/home/user/mbzirc_ws/install/lib/libGameLogicPlugin.so", at 0x7f4f1db2aa70, in void std::_Bind<void (mbzirc::GameLogicPluginPrivate::*(mbzirc::GameLogicPluginPrivate*))()>::operator()<, void>()
[ign gazebo-1] #7    Object "/home/user/mbzirc_ws/install/lib/libGameLogicPlugin.so", at 0x7f4f1db3818e, in void std::_Bind<void (mbzirc::GameLogicPluginPrivate::*(mbzirc::GameLogicPluginPrivate*))()>::__call<void, , 0ul>(std::tuple<>&&, std::_Index_tuple<0ul>)
[ign gazebo-1] #6    Object "/home/user/mbzirc_ws/install/lib/libGameLogicPlugin.so", at 0x7f4f1db45df2, in std::__invoke_result<void (mbzirc::GameLogicPluginPrivate::*&)(), mbzirc::GameLogicPluginPrivate*&>::type std::__invoke<void (mbzirc::GameLogicPluginPrivate::*&)(), mbzirc::GameLogicPluginPrivate*&>(void (mbzirc::GameLogicPluginPrivate::*&)(), mbzirc::GameLogicPluginPrivate*&)
[ign gazebo-1] #5    Object "/home/user/mbzirc_ws/install/lib/libGameLogicPlugin.so", at 0x7f4f1db526e7, in void std::__invoke_impl<void, void (mbzirc::GameLogicPluginPrivate::*&)(), mbzirc::GameLogicPluginPrivate*&>(std::__invoke_memfun_deref, void (mbzirc::GameLogicPluginPrivate::*&)(), mbzirc::GameLogicPluginPrivate*&)
[ign gazebo-1] #4    Object "/home/user/mbzirc_ws/install/lib/libGameLogicPlugin.so", at 0x7f4f1daf937c, in mbzirc::GameLogicPluginPrivate::OnPostRender()
[ign gazebo-1] #3    Object "/home/user/mbzirc_ws/install/lib/libGameLogicPlugin.so", at 0x7f4f1daf9e7a, in mbzirc::GameLogicPluginPrivate::SaveImage(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
[ign gazebo-1] #2    Object "/usr/lib/x86_64-linux-gnu/ign-rendering-6/engine-plugins/libignition-rendering-ogre2.so", at 0x7f4ee9c8e908, in ignition::rendering::v6::BaseCamera<ignition::rendering::v6::Ogre2Sensor>::Copy(ignition::rendering::v6::Image&) const
[ign gazebo-1] #1    Object "/usr/lib/x86_64-linux-gnu/ign-rendering-6/engine-plugins/libignition-rendering-ogre2.so", at 0x7f4ee9d1c623, in ignition::rendering::v6::Ogre2RenderTarget::Copy(ignition::rendering::v6::Image&) const
[ign gazebo-1] #0    Object "/lib/x86_64-linux-gnu/libOgreMain.so.2.2.6", at 0x7f4ee98e0d64, in Ogre::TextureGpu::getPixelFormat() const
[ign gazebo-1] Segmentation fault (Address not mapped to object [0x54])
[video_target_relay-48] [INFO] [1660397996.726578913] [quadrotor_17.video_target_relay]: Save target report image to /home/user/.ros/mbzirc/vessel.png.
[ign gazebo-1] [Dbg] [Gpu1;32mignition-rendering-ogre2]
[ign gazebo-1] [GUI] [Dbg] [SelectEntities.cc:451] SelectEntities plugin is using camera [scene::Camera(65527)]
[ign gazebo-1] [GUI] [Msg] Loading plugin [ignition-rendering-ogre2]
[ign gazebo-1] [GUI] [Dbg] [MarkerManager.cc:171] Advertise /marker/list service.
[ign gazebo-1] [GUI] [Dbg] [MarkerManager.cc:181] Advertise /marker/list.
[ign gazebo-1] [GUI] [Dbg] [MarkerManager.cc:191] Advertise /marker_array.
[ign gazebo-1] [GUI] [Msg] Loading plugin [ignition-rendering-ogre2]
[ign gazebo-1] [GUI] [Dbg] [CameraTracking.cc:181] CameraTrackingPrivate plugin is moving camera [scene::Camera(65527)]
[ign gazebo-1] [GUI] [Msg] Move to service on [/gui/move_to]
[ign gazebo-1] [GUI] [Msg] Follow service on [/gui/follow]
[ign gazebo-1] [GUI] [Msg] Move to pose service on [/gui/move_to/pose]
[ign gazebo-1] [GUI] [Msg] Camera pose topic advertised on [/gui/camera/pose]
[ign gazebo-1] [GUI] [Msg] Follow offset service on [/gui/follow/offset]
[ign gazebo-1] [GUI] [Msg] Loading plugin [ignition-rendering-ogre2]
[ign gazebo-1] [GUI] [Dbg] [InteractiveViewControl.cc:130] InteractiveViewControl plugin is moving camera [scene::Camera(65527)]
[ign gazebo-1] [GUI] [Msg] Loading plugin [ignition-rendering-ogre2]
[ign gazebo-1] [GUI] [Dbg] [Ogre2ParticleEmitter.cc:514] Particle emitter initialized
[ign gazebo-1] [GUI] [Dbg] [Ogre2ParticleEmitter.cc:514] Particle emitter initialized
[ign gazebo-1] [GUI] [Dbg] [Ogre2ParticleEmitter.cc:514] Particle emitter initialized
[ign gazebo-1] [GUI] [Dbg] [Ogre2ParticleEmitter.cc:514] Particle emitter initialized
[ign gazebo-1] [GUI] [Dbg] [Ogre2ParticleEmitter.cc:514] Particle emitter initialized
[ign gazebo-1] [GUI] [Dbg] [Ogre2ParticleEmitter.cc:514] Particle emitter initialized
[ign gazebo-1] [GUI] [Msg] Loading plugin [ignition-rendering-ogre2]
[ign gazebo-1] [GUI] [Msg] Loading plugin [ignition-rendering-ogre2]
[ign gazebo-1] [GUI] [Msg] Loading plugin [ignition-rendering-ogre2]
[ign gazebo-1] [GUI] [Msg] Loading plugin [ignition-rendering-ogre2]
[ign gazebo-1] [GUI] [Msg] Loading plugin [ignition-rendering-ogre2]
[ign gazebo-1] [GUI] [Dbg] [EntityContextMenuPlugin.cc:79] Entity context menu plugin is using camera [scene::Camera(65527)]
[ign gazebo-1] [GUI] [Dbg] [SignalHandler.cc:141] Received signal[2].
[ign gazebo-1] [GUI] [Dbg] [Gui.cc:331] Shutting down ign-gazebo-gui
[ign gazebo-1] [GUI] [Dbg] [Application.cc:134] Terminating application.
[ign gazebo-1] [GUI] [Msg] Loading plugin [ignition-rendering-ogre2]
[ign gazebo-1] [GUI] [Dbg] [MinimalScene.cc:583] Destroy scene [scene]
[INFO] [ign gazebo-1]: process has finished cleanly [pid 618899]
[INFO] [monitor_sim-2]: process has finished cleanly [pid 618902]
[INFO] [video_target_relay-97]: sending signal 'SIGINT' to process[video_target_relay-97]
[INFO] [pose_tf_broadcaster-96]: sending signal 'SIGINT' to process[pose_tf_broadcaster-96]
[INFO] [parameter_bridge-95]: sending signal 'SIGINT' to process[parameter_bridge-95]
[INFO] [video_target_relay-94]: sending signal 'SIGINT' to process[video_target_relay-94]
[... 239 lines cut ...]
[INFO] [parameter_bridge-55]: sending signal 'SIGTERM' to process[parameter_bridge-55]
[INFO] [parameter_bridge-58]: process has finished cleanly [pid 619760]
[INFO] [parameter_bridge-52]: sending signal 'SIGTERM' to process[parameter_bridge-52]
[INFO] [parameter_bridge-55]: process has finished cleanly [pid 619720]
[INFO] [parameter_bridge-49]: sending signal 'SIGTERM' to process[parameter_bridge-49]
[INFO] [parameter_bridge-52]: process has finished cleanly [pid 619697]
[INFO] [parameter_bridge-46]: sending signal 'SIGTERM' to process[parameter_bridge-46]
[INFO] [parameter_bridge-49]: process has finished cleanly [pid 619645]
[INFO] [parameter_bridge-41]: sending signal 'SIGTERM' to process[parameter_bridge-41]
[INFO] [parameter_bridge-46]: process has finished cleanly [pid 619597]
[INFO] [parameter_bridge-38]: sending signal 'SIGTERM' to process[parameter_bridge-38]
[INFO] [parameter_bridge-41]: process has finished cleanly [pid 619482]
[INFO] [parameter_bridge-35]: sending signal 'SIGTERM' to process[parameter_bridge-35]
[INFO] [parameter_bridge-32]: sending signal 'SIGTERM' to process[parameter_bridge-32]
[INFO] [parameter_bridge-38]: process has finished cleanly [pid 619443]
[INFO] [parameter_bridge-35]: process has finished cleanly [pid 619428]
[... cut ...] ' to process[parameter_bridge-29]
[INFO] [parameter_bridge-32]: process has finished cleanly [pid 619413]
[INFO] [parameter_bridge-26]: sending signal 'SIGTERM' to process[parameter_bridge-26]
[INFO] [parameter_bridge-25]: sending signal 'SIGTERM' to process[parameter_bridge-25]
[INFO] [parameter_bridge-29]: process has finished cleanly [pid 619398]
[INFO] [parameter_bridge-26]: process has finished cleanly [pid 619383]
[INFO] [parameter_bridge-25]: process has finished cleanly [pid 619241]

```

iche033, 2022-08-15:
Can you check if you have ignition-rendering version 6.5.1?

jiazk1226, 2022-08-16:
Hello, we are facing the same problem. In the previous [issue,](https://github.com/osrf/mbzirc/issues/193) we planned to use different cameras of different UAVs to report the positions of the target vessel and the object respectively. However, in the new version, after we received `vessel_id_success`, we close the streaming, then use another UAV to identify the object and report the position. When the object is identified and reported, the process of mbzirc_simulater shut down and reports the following error message. However, when we use the same camera on the same UAV, everything goes well. We can get feedback from `/mbzirc/phase` in turn: `vessel_id_success >> small_object_id_success >> large_object_id_success`.

We checked the ignition-rendering version and the version is 6.5.1.

Thank you.

 ```
[ign gazebo-1] ign gazebo server: /var/lib/jenkins/workspace/ogre-2.2-debbuilder/repo/OgreMain/src/OgreTextureGpu.cpp:175: void Ogre::TextureGpu::setResolution(Ogre::uint32, Ogre::uint32, Ogre::uint32): Assertion `mResidencyStatus == GpuResidency::OnStorage || isRenderWindowSpecific()' failed.
[ign gazebo-1] Stack trace (most recent call last) in thread 2335238:
[ign gazebo-1] #22   Object "[0xffffffffffffffff]", at 0xffffffffffffffff, in 
[ign gazebo-1] #21   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7fd884ab2132, in clone
[ign gazebo-1] #20   Object "/lib/x86_64-linux-gnu/libpthread.so.0", at 0x7fd884978608, in 
[ign gazebo-1] #19   Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7fd8807a8de3, in 
[ign gazebo-1] #18   Object "/usr/lib/x86_64-linux-gnu/ign-gazebo-6/plugins/libignition-gazebo-sensors-system.so", at 0x7fd856afb867, in ignition::gazebo::v6::systems::SensorsPrivate::RenderThread()
[ign gazebo-1] #17   Object "/usr/lib/x86_64-linux-gnu/ign-gazebo-6/plugins/libignition-gazebo-sensors-system.so", at 0x7fd856afb1f2, in ignition::gazebo::v6::systems::SensorsPrivate::RunOnce()
[ign gazebo-1] #16   Object "/home/vision/mbzirc_ws/install/lib/libGameLogicPlugin.so", at 0x7fd854670ff4, in std::_Function_handler<void (), std::_Bind<void (mbzirc::GameLogicPluginPrivate::*(mbzirc::GameLogicPluginPrivate*))()> >::_M_invoke(std::_Any_data const&)
[ign gazebo-1] #15   Object "/home/vision/mbzirc_ws/install/lib/libGameLogicPlugin.so", at 0x7fd85467f2fe, in void std::_Bind<void (mbzirc::GameLogicPluginPrivate::*(mbzirc::GameLogicPluginPrivate*))()>::operator()<, void>()
[ign gazebo-1] #14   Object "/home/vision/mbzirc_ws/install/lib/libGameLogicPlugin.so", at 0x7fd85468cdd0, in void std::_Bind<void (mbzirc::GameLogicPluginPrivate::*(mbzirc::GameLogicPluginPrivate*))()>::__call<void, , 0ul>(std::tuple<>&&, std::_Index_tuple<0ul>)
[ign gazebo-1] #13   Object "/home/vision/mbzirc_ws/install/lib/libGameLogicPlugin.so", at 0x7fd85469abfa, in std::__invoke_result<void (mbzirc::GameLogicPluginPrivate::*&)(), mbzirc::GameLogicPluginPrivate*&>::type std::__invoke<void (mbzirc::GameLogicPluginPrivate::*&)(), mbzirc::GameLogicPluginPrivate*&>(void (mbzirc::GameLogicPluginPrivate::*&)(), mbzirc::GameLogicPluginPrivate*&)
[ign gazebo-1] #12   Object "/home/vision/mbzirc_ws/install/lib/libGameLogicPlugin.so", at 0x7fd8546a7697, in void std::__invoke_impl<void, void (mbzirc::GameLogicPluginPrivate::*&)(), mbzirc::GameLogicPluginPrivate*&>(std::__invoke_memfun_deref, void (mbzirc::GameLogicPluginPrivate::*&)(), mbzirc::GameLogicPluginPrivate*&)
[ign gazebo-1] #11   Object "/home/vision/mbzirc_ws/install/lib/libGameLogicPlugin.so", at 0x7fd85464c642, in mbzirc::GameLogicPluginPrivate::OnPostRender()
[ign gazebo-1] #10   Object "/home/vision/mbzirc_ws/install/lib/libGameLogicPlugin.so", at 0x7fd85464d12a, in mbzirc::GameLogicPluginPrivate::FindTargetVisual(std::shared_ptr<ignition::rendering::v6::Visual> const&, ignition::math::v6::Vector2<int> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&) const
[ign gazebo-1] #9    Object "/home/vision/mbzirc_ws/install/lib/libGameLogicPlugin.so", at 0x7fd85464e151, in mbzirc::GameLogicPluginPrivate::VisualAt(unsigned int, unsigned int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const
[ign gazebo-1] #8    Object "/usr/lib/x86_64-linux-gnu/ign-rendering-6/engine-plugins/libignition-rendering-ogre2.so", at 0x7fd830203c95, in ignition::rendering::v6::Ogre2Camera::VisualAt(ignition::math::v6::Vector2<int> const&)
[ign gazebo-1] #7    Object "/usr/lib/x86_64-linux-gnu/ign-rendering-6/engine-plugins/libignition-rendering-ogre2.so", at 0x7fd8302029f3, in ignition::rendering::v6::Ogre2Camera::SetSelectionBuffer()
[ign gazebo-1] #6    Object "/usr/lib/x86_64-linux-gnu/ign-rendering-6/engine-plugins/libignition-rendering-ogre2.so", at 0x7fd8302a3a6f, in ignition::rendering::v6::Ogre2SelectionBuffer::Ogre2SelectionBuffer(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::shared_ptr<ignition::rendering::v6::Ogre2Scene>, unsigned int, unsigned int)
[ign gazebo-1] #5    Object "/usr/lib/x86_64-linux-gnu/ign-rendering-6/engine-plugins/libignition-rendering-ogre2.so", at 0x7fd8302a2827, in ignition::rendering::v6::Ogre2SelectionBuffer::CreateRTTBuffer()
[ign gazebo-1] #4    Object "/lib/x86_64-linux-gnu/libOgreMain.so.2.2.6", at 0x7fd8148b019f, in Ogre::TextureGpu::setResolution(unsigned int, unsigned int, unsigned int)
[... cut ...]
6.4.0-1~focal - libignition-rendering6-core-dev (5 6.4.0-1~focal) libignition-rendering6-ogre1-dev (5 6.4.0-1~focal) libignition-rendering6-ogre2-dev (5 6.4.0-1~focal) libignition-rendering6 (5 6.4.0-1~focal) 
6.3.1-1~focal - libignition-rendering6-core-dev (5 6.3.1-1~focal) libignition-rendering6-ogre1-dev (5 6.3.1-1~focal) libignition-rendering6-ogre2-dev (5 6.3.1-1~focal) libignition-rendering6 (5 6.3.1-1~focal) 
Provides: 
6.5.1-1~focal - 
6.4.0-1~focal - 
6.3.1-1~focal - 
Reverse Provides:
```

iche033, 2022-08-16:
Can you also check: `libignition-rendering6-ogre2-dev`, e.g. post the output of

```
dpkg -l | grep libignition-rendering6-ogre2-dev
```

In your `mbzirc_ws`, you can also double check and verify that it's linking against the correct ign-rendering library

```
ldd `find install/ | grep GameLogic` | grep rendering6
```

The process we have been using for testing switching streams when reporting target is described in #195. Let me know if you see anything different that we're doing from your use case.

jiazk1226, 2022-08-17:
Thank you, this solved my problem!
I input `dpkg -l | grep libignition-rendering6-ogre2-dev` in the terminal and find the version of `libignition-rendering6-ogre2-dev` is 6.4.1. Then I update it and everything was done.
Thank you again!

iche033, 2022-08-17:
great to hear that!

iche033, 2022-08-17:
@chalseokim looking closely at your backtrace, it seems to be related to a recent change added for logging target report images. To be safe, I disabled it in https://github.com/osrf/mbzirc/pull/214.

dongk-97, 2022-08-17:
We have checked the version of the ignition rendering and it is 6.5.1. Also, we checked that the linking is correct as below.
```
dpkg -l | grep libignition-rendering6-ogre2-dev
ii  libignition-rendering6-ogre2-dev:amd64              6.5.1-1~focal                         amd64        Ignition rendering classes and functions for robot apps - Development files
```
We found that the phase transition works fine when we use vga camera and the shutting down issue occurs when the endpoint camera is an rgbd. 

Are we not allowed to stream using an rgbd or is this a bug? If not, do we have a different dependency issue?

Thank you.

iche033, 2022-08-17:
> Are we not allowed to stream using an rgbd or is this a bug? If not, do we have a different dependency issue?

Found that it's a bug. Do you need to use rgbd camera for reporting targets or can you workaround the issue? In any case, I have created a PR to fix the issue in https://github.com/osrf/mbzirc/pull/216 first.

closed 2022-08-17 by iche033