osrf/mbzirc #211: RGBD camera points cloud

fbenti, 2022-08-16:
Hi,

I'm trying to get the point clouds given by the depth camera, however I'm finding trouble extracting the rgb values correctly.

I'm subscribing to the /quadrotor_1/slot3/points topic. In the respective PointField[] fields there is a description of the channels and their layout in the binary data blob (as shown in the picture below).
[image]

The 'point_step' field in the PointCloud2 message is equal to 24, meaning that each point cloud is represented by 24 bytes. Considering the description in the picture, i would expect that the x-y-z value are 4 bytes each and the remaining 12 are given by r-g-b values. Hence, looking at the last channel, 'rgb', I would expect it to start at offset 12 and having count = 3, meaning that there are 4 bytes for each channel, instead  of having offset=16 and count = 1.

iche033, 2022-08-16:
the rgb values are 1 byte per channel. They are packed into 4 byte data type (7). So the first byte is `r`, second byte is `g`, etc.

fbenti, 2022-08-16:
Yeah that was my guess, however since the point_step is 24 (bytes) and the rgb channel starts at offset 16, there should be something wrong.

iche033, 2022-08-16:
I think the `rgb` value could in theory start at offset 12 and occupy the next 4 bytes. We won't change the behavior now though. In any case, the each point cloud occupies 16 out of 24 bytes and the rest are just padding. You'll just need to get the rgb values from offset 16 instead of offset 12.

closed 2022-08-18 by fbenti