osrf/mbzirc #215: Problems about Submission

xirhxq, 2022-08-17:
Hello Sir,
We are facing a problem during submission. When we try to use AWS to submit the docker image, the output of submission tell that the upload failed by "An error occurred (AccessDenied) when calling the PutObject operation: Access Denied". Amazon's official explanation for this issue is as follows: "Your users need to have the s3;PutObject permission." Could you please tell us how to solve this problem?

Thanks.

iche033, 2022-08-17:
hi, what team are you on? I'll verify your credentials

xirhxq, 2022-08-18:
hi, our team is 'fly_eagle', which is the parameter when running 'bash mbzirc_submit.bash', if that's what is needed.

By the way, by 23rd Aug, is it an upper limit of submits we can do? What is the maximum times we can submit?

iche033, 2022-08-18:
hi, the team name should be `fly-eagle` with a `-`. I tested your credentials and they should work.

> By the way, by 23rd Aug, is it an upper limit of submits we can do? What is the maximum times we can submit?

Yes. 23rd is the last day for submitting dry run.  Each time you submit, it's going overwrite the previous one. So you can re-submit if something went wrong. Each submission does not automatically trigger a run. We will be manually launching and running your solution during dry-run. If we find anything wrong with the run, we'll get in touch with you. If you are able to fix the issue before dry run period ends, we can potentially run it again.

xirhxq, 2022-08-18:
Hi, thanks a lot for the reply! We will try to submit as early as possible.

LvJohny, 2022-08-18:
Hi, we are the same team. We have tried change the team name to fly-eagle. But the problem still exists by "An error occurred (AccessDenied) when calling the PutObject operation: Access Denied".

iche033, 2022-08-18:
I have just re-sent the credentials to the main and alternative contact emails addresses we have. Please check in case they are different from the ones sent before. I have also just tested submitting a solution using your credentials and I was able to submit. Make sure when configuring aws command line tool, you're using `us-east-1` region.

LvJohny, 2022-08-19:
Hi, thanks a lot for the reply!By using the credentials you provide us, we are able to submit the solution.

iche033, 2022-08-19:
thanks I see your submissions. For dry-run, can you submit with the upstream tag `dry_run`? Currently I see other tags being used.

xirhxq, 2022-08-20:
Definitely Ian, we will set the upstream tag only `dry_run`. For now we have done a new submission. It's not the whole solution, but it contains every dependencies we need. We'll appreciate it if you  could give it a run test. Thanks.