osrf/mbzirc #218: aws only returns yaml as output of submission

fjdk99, 2022-08-22:
[image]

We tried to submit our docker image for the test/dry run. However, the aws only returns the team configuration yaml file as the output when we typed "aws s3 sync s3://cloudsim-mbzirc-logs/team-kaist/ logs". 

We actually checked the nodes for each vehicles are running with ./run_solution.bash <vehicle name>.
[image]

How can we receive the full logs from aws? Is there any waiting time for producing the logs from aws server?

iche033, 2022-08-22:
The runs are triggered manually during dry run. We will be running your solution today.

pine-s, 2022-08-23:
What's the timezone for the due date? Aug. 23rd of PST or GST?