osrf/mbzirc #221: Image "osrf/mbzirc:cloudsim_bridge_latest" does not contains cycloneadds.xml

pine-s, 2022-08-23:
[image]
 We have found that the bridges executed from the docker compose died immediately because 'cycloneadds.xml' is not found in the "osrf/mbzirc:cloudsim_bridge_latest".

We've found modifying the image "osrf/mbzirc:cloudsim_bridge_latest" with dockerfile could solve the issue, and run our USV and drones as we intended. We want to check whether the previous test run was done without the shutdown of the bridge containers.
```
FROM osrf/mbzirc:mbzirc_sim_latest

RUN mkdir /home/developer/config
COPY run_bridge.bash ./
COPY cyclonedds.xml /home/developer/config/
ENTRYPOINT ["./run_bridge.bash"]
```

closed 2022-08-23 by pine-s

pieniacy, 2022-08-23:
We experienced the exact same problem when we tried to run the docker compose setup on our infrastructure. Can we discuss this problem in this issue, so it's publicly available to everyone?

iche033, 2022-08-23:
Posting my answer here in case it helps others. The original issue is about missing `cyclonedds.xml` in the bridge container.

If you don't see the `cyclonedds.xml` file in the bridge container, it could be that docker compose is not run from inside the `docker/compose` directory. There is a [cyclonedds.xml](https://github.com/osrf/mbzirc/blob/v4/docker/compose/cyclonedds.xml) file in that directory which gets mounted to the container by [this line](https://github.com/osrf/mbzirc/blob/v4/docker/compose/gen_docker_compose.py#L106). It uses the `.` current path. You can try changing this path to the absolute path of the `docker/compose` directory containing the `cyclonedds.xml` file.