osrf/mbzirc #222: Crash after reporting target

fbenti, 2022-08-23:
After reporting the target vessel back to the operator, gazebo crashes unexpectedly but we can't understand why.

> [ign gazebo-1] [Msg] Target stream start request: success
[ign gazebo-1] [Msg] Target report in stream: vessel 322 154
[ign gazebo-1] Stack trace (most recent call last) in thread 779977:
[ign gazebo-1] #15   Object "[0xffffffffffffffff]", at 0xffffffffffffffff, in 
[ign gazebo-1] #14   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7fc6be867132, in clone
[ign gazebo-1] #13   Object "/lib/x86_64-linux-gnu/libpthread.so.0", at 0x7fc6be72d608, in 
[ign gazebo-1] #12   Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7fc6ba877de3, in 
[ign gazebo-1] #11   Object "/usr/lib/x86_64-linux-gnu/ign-gazebo-6/plugins/libignition-gazebo-sensors-system.so", at 0x7fc694bc4867, in ignition::gazebo::v6::systems::SensorsPrivate::RenderThread()
[ign gazebo-1] #10   Object "/usr/lib/x86_64-linux-gnu/ign-gazebo-6/plugins/libignition-gazebo-sensors-system.so", at 0x7fc694bc41f2, in ignition::gazebo::v6::systems::SensorsPrivate::RunOnce()
[ign gazebo-1] #9    Object "/home/aut/mbzirc_ws/install/lib/libGameLogicPlugin.so", at 0x7fc69272eff4, in std::_Function_handler<void (), std::_Bind<void (mbzirc::GameLogicPluginPrivate::*(mbzirc::GameLogicPluginPrivate*))()> >::_M_invoke(std::_Any_data const&)
[ign gazebo-1] #8    Object "/home/aut/mbzirc_ws/install/lib/libGameLogicPlugin.so", at 0x7fc69273d2fe, in void std::_Bind<void (mbzirc::GameLogicPluginPrivate::*(mbzirc::GameLogicPluginPrivate*))()>::operator()<, void>()
[ign gazebo-1] #7    Object "/home/aut/mbzirc_ws/install/lib/libGameLogicPlugin.so", at 0x7fc69274add0, in void std::_Bind<void (mbzirc::GameLogicPluginPrivate::*(mbzirc::GameLogicPluginPrivate*))()>::__call<void, , 0ul>(std::tuple<>&&, std::_Index_tuple<0ul>)
[ign gazebo-1] #6    Object "/home/aut/mbzirc_ws/install/lib/libGameLogicPlugin.so", at 0x7fc692758bfa, in std::__invoke_result<void (mbzirc::GameLogicPluginPrivate::*&)(), mbzirc::GameLogicPluginPrivate*&>::type std::__invoke<void (mbzirc::GameLogicPluginPrivate::*&)(), mbzirc::GameLogicPluginPrivate*&>(void (mbzirc::GameLogicPluginPrivate::*&)(), mbzirc::GameLogicPluginPrivate*&)
[ign gazebo-1] #5    Object "/home/aut/mbzirc_ws/install/lib/libGameLogicPlugin.so", at 0x7fc692765697, in void std::__invoke_impl<void, void (mbzirc::GameLogicPluginPrivate::*&)(), mbzirc::GameLogicPluginPrivate*&>(std::__invoke_memfun_deref, void (mbzirc::GameLogicPluginPrivate::*&)(), mbzirc::GameLogicPluginPrivate*&)
[ign gazebo-1] #4    Object "/home/aut/mbzirc_ws/install/lib/libGameLogicPlugin.so", at 0x7fc69270a252, in mbzirc::GameLogicPluginPrivate::OnPostRender()
[ign gazebo-1] #3    Object "/home/aut/mbzirc_ws/install/lib/libGameLogicPlugin.so", at 0x7fc69270ad50, in mbzirc::GameLogicPluginPrivate::SaveImage(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
[ign gazebo-1] #2    Object "/usr/lib/x86_64-linux-gnu/ign-rendering-6/engine-plugins/libignition-rendering-ogre2.so", at 0x7fc66a1f0908, in ignition::rendering::v6::BaseCamera<ignition::rendering::v6::Ogre2Sensor>::Copy(ignition::rendering::v6::Image&) const
[ign gazebo-1] #1    Object "/usr/lib/x86_64-linux-gnu/ign-rendering-6/engine-plugins/libignition-rendering-ogre2.so", at 0x7fc66a27e623, in ignition::rendering::v6::Ogre2RenderTarget::Copy(ignition::rendering::v6::Image&) const
[ign gazebo-1] #0    Object "/lib/x86_64-linux-gnu/libOgreMain.so.2.2.6", at 0x7fc669e42d64, in Ogre::TextureGpu::getPixelFormat() const
[ign gazebo-1] Segmentation fault (Address not mapped to object [0x54])
[ign gazebo-1] [GUI] [Dbg] [Ogre2ParticleEmitter.cc:514] Particle emitter initialized
[ign gazebo-1] [GUI] [Dbg] [Ogre2ParticleEmitter.cc:514] Particle emitter initialized
[ign gazebo-1] [GUI] [Dbg] [Ogre2ParticleEmitter.cc:514] Particle emitter initialized
[ign gazebo-1] [GUI] [Dbg] [Ogre2ParticleEmitter.cc:514] Particle emitter initialized
[ign gazebo-1] [GUI] [Dbg] [Ogre2ParticleEmitter.cc:514] Particle emitter initialized
[ign gazebo-1] [GUI] [Dbg] [Ogre2ParticleEmitter.cc:514] Particle emitter initialized
[ign gazebo-1] [GUI] [Msg] Loading plugin [ignition-rendering-ogre2]
[ign gazebo-1] [GUI] [Dbg] [SignalHandler.cc:141] Received signal[2].
[ign gazebo-1] [GUI] [Dbg] [Gui.cc:331] Shutting down ign-gazebo-gui
[ign gazebo-1] [GUI] [Dbg] [Application.cc:134] Terminating application.
[ign gazebo-1] [GUI] [Msg] Loading plugin [ignition-rendering-ogre2]
[ign gazebo-1] [GUI] [Dbg] [MinimalScene.cc:583] Destroy scene [scene]
[INFO] [ign gazebo-1]: process has finished cleanly [pid 779785]

iche033, 2022-08-23:
could be related to: https://github.com/osrf/mbzirc/issues/209

If you are using an rgbd camera for reporting targets, you'll need to update to the latest version that has the fix from https://github.com/osrf/mbzirc/pull/216. If you are switching streams when reporting targets, you'll need to make sure you have `libignition-rendering6-ogre2-dev` 6.5.1

fbenti, 2022-08-23:
I can actually see that the GameLogicPlugin.cc in its latest version on GitHub is different from our local version, even if we just reinstalled everything. How can we solve this problem?

iche033, 2022-08-23:
It depends. If you're using docker, you can remove the current image (, e.g. `docker rmi -f`) and pull it again just to make sure you have the latest version.

If you're building from source, just checkout the `main` branch and `git pull`