osrf/mbzirc #223: IMU or barometer does't send any data

LvJohny, 2022-08-24:
Hi. In our tests, we found that in separate tests, the onboard sensors of the UAV sometimes do not send any data, including barometer and IMU. When we restart the emulator, the sensors will send data and return to normal. If the sensor does not work, our system will not be able to start. We want to know whether this is intentionally set by the official? If not, whether additional test opportunities will be obtained in the final evaluation due to sensor failure.

xirhxq, 2022-08-24:
When we integrate all our codes and generate 20 UAVs, the probability that IMU will not send data greatly increases. We are deeply trapped. Could you help us? Is there anything to avoid this? The following figure is a screenshot of a time when we used rqt to monitor the frequency of IMU. Whether we restart the simulator or the computer, there is no IMU data of our suav_7.

[image]
.

iche033, 2022-08-24:
hi, checking see if this is all from running on the same machine? Does this happen only when there are many (20) vehicles spawned in simulation.  wonder if it's CPU bound. The bridge container and its processes may be contending for resources.

LvJohny, 2022-08-25:
>checking see if this is all from running on the same machine? 

All our code runs on the same machine. We first launch the simulator with config_team.yaml, and then run our code. We find this problem because sometimes our one UAV do not take off or fly to the wrong altitude. Later, after we started the simulator, we directly checked the imu of UAVs and determined that it was a random problem when starting the simulator. 

>Does this happen only when there are many (20) vehicles spawned in simulation. 

No, in the previous 7 drone tests, this also happened but with a low probability. We can always restart the simulator to solve this problem with an affordable time cost.

>wonder if it's CPU bound

Our cpu is i9-12900k. When we only start the simulator and generate UAVs, our cpu usage rate is only 20% to 25%, but it sometimes happens.

iche033, 2022-08-26:
Some updates: We did a few more runs internally on our cloud simulation platform and we have not seen the issue happen yet inside the solution containers on the cloud machines. We tested with different teams solutions. It does not mean the issue does not exist. It could be that it's harder to reproduce on the cloud. When running the final evaluation, we will check to make sure that data are coming through after launching simulation. If we do see the issue with missing data on the ROS2 topics, we will kill the run and start a new one.

LvJohny, 2022-08-26:
Ok, Thank you for your reply and work。

closed 2022-08-26 by LvJohny