osrf/mbzirc #226: New light gripper causes stability issues...

Fergus-MW, 2022-08-30:
Hi There,

Over the last few days, I have been trying out the new gripper (mbzirc_suction_gripper_light) after seeing it mentioned in some of the threads here. With the regular suction gripper, we are able to reliably attach to the small package, but can then not lift it, so we thought it might be worth trying this new one in case it enables the drone to take off. After installing it however, the drone is definitely less stable, as a very high control loop frequency is required to calm the oscillations it exhibits when trying to home in on the small package. Such a high control loop frequency is not available when 10+ drones are being simulated at once, as the CPU (i7 12700k) is maxed out. With the old gripper, we were able to maintain stable control of the drones with a control loop frequency of around 4Hz, now 20 Hz minimum is required to guide the drone to the package. Is there a known solution to this problem? We have made considerable efforts to adapt our control systems to cater to the new gripper, but could not find a solution that works anywhere near as reliably as it did with the old gripper.

Best wishes,

Fergus

iche033, 2022-08-30:
hi, when we're running your solution on our cloud simulation platform, we'll deploy max of 2 solution containers per [g3.4xlarge](https://aws.amazon.com/ec2/instance-types/g3/) machine instance on AWS. The simulator runs separately on a different machine. So in theory you'll get more CPU resources to run your control algorithm. 

Also checking if 20Hz is measured in simulation time or wall clock time? From what we observed, running ~20 drones on the cloud gives roughly somewhere between 0.1 ~ 0.2 real time factor in simulation, i.e. 1/10th ~ 1/5th of the speed of real time. So if the control loop frequency measurement given is in simulation time, the freq needed in real time should be lower, around (20 * 0.1) 2Hz ~ (20 * 0.2) 4Hz. However, timing issues are tricky and I am not able guarantee if that will allow you to hit the target control loop frequency though.

Fergus-MW, 2022-08-30:
Hi Ian,

Thanks for getting back to me. Even with 20Hz control loops, it is still nowhere near as reliable as the drone was with the old gripper. Is there any way we can have the stable manoeuvring of the old gripper whilst being able to lift the small package?

Best wishes,

Fergus

iche033, 2022-08-30:
hi, I'm afraid I'm not able to make changes on our end at this point. We have pinned the version and ready for final evaluation in a couple of days. The organizer made an exception to get the lightweight version of the gripper in just before feature freeze provided that the changes are minimal and non-breaking. So we did not make more changes to try and tune the flight stability. You probably have found this already but more info can be found in https://github.com/osrf/mbzirc/pull/196. 

The old suction gripper requires very accurate placement of the suction device at the center of mass of the small object while the lightweight version requires better flight control so both have their own caveats / challenges.  I think you would need to try with higher control freq and/or slower drone flight maneuvers.