osrf/mbzirc #228: Issues when reporting target stream

fzoric8, 2022-09-06:
Hi, 

We've been running reduced simulation to record video of whole simulation. 

We've been following instructions found on wiki and we get following output in our `events.yaml`: 
```
event:
  id: 1
  type: stream_start_request
  time_sec: 31
  elapsed_real_time: 135
  elapsed_sim_time: 18
  total_score: 18
  data: success
- event:
  id: 2
  type: target_reported_in_stream
  time_sec: 264
  elapsed_real_time: 1629
  elapsed_sim_time: 251
  total_score: 251
  data: vessel::516::332
```
Stream has started and vessel is detected, however, after 3 detections as this one (correct pixel and correct vessel) we get `vessel_id_failure`. 

[image]

I'm providing screenshot for convenience. 

If we compare image saved in logs and stuff logged in `events.yaml`, everything seems fine, what else could be a problem regarding `vessel_id_failure`? 

Thank you in advance.

iche033, 2022-09-06:
hi from the logs, they all look fine. The example `coast.sdf` world has vessel B as the target object. If you're recording videos with Vessel E, just checking if you have set the target vessel  to be Vessel E [here](https://github.com/osrf/mbzirc/blob/v4/mbzirc_ign/worlds/coast.sdf#L1457-L1463)?

You can also see more info printed in the console about target reports that may help debug the issue. The msgs maybe bufferred though and may not be prnited out right away. In that case, once the simulation is paused after 3 failed reports, play the simulation again, run command `ign topic -t "/Vessel_B/detach" -m ignition.msgs.Empty -p "unused: true"` to force the simulator to printout more msgs. You should then see information like below that says which target vessel is being set and where it is in the image:

```
[ign gazebo-1] [Msg] Target report in stream: vessel 640 40
[ign gazebo-1] [Msg] Loading plugin [ignition-rendering-ogre2]
[ign gazebo-1] [Msg] No valid target at (640 40).
[ign gazebo-1] [Msg] Checking nearby pixels.
[ign gazebo-1] [Msg] Target is in view: Vessel B at 640 487
[ign gazebo-1] [Msg]   Target is not within the allowed tolerance: diff: 447, tol: 38.4
```

fzoric8, 2022-09-07:
You're right. Vessel B was configured as default object in coast world. 
Thank you very much :)

closed 2022-09-07 by fzoric8