osrf/mbzirc #230: Requsting List of world names

Walid-khaled, 2022-12-07:
after finishing setup and running the simulation, this problem arises.
[image]

iche033, 2022-12-08:
Are you using source build or docker? What command did you run to launch simulation?

Walid-khaled, 2022-12-09:
I tried both and I got the same error. The command I tried as mentioned here: https://github.com/osrf/mbzirc/wiki/Running-the-Demo

for example, 
source install/setup.bash
ros2 launch mbzirc_ign spawn.launch.py name:=usv world:=simple_demo model:=usv type:=usv x:=15 y:=0 z:=0.3 R:=0 P:=0 Y:=0

iche033, 2022-12-10:
I'll need to see more of the console output to know what the issue is.

Can you try just launching a simple gazebo world without MBZIRC and see if the simulator is working for you? e.g.

```
ign gazebo -v 4 shapes.sdf
```

or

```
ign gazebo -v 4 tunnel.sdf
```

Walid-khaled, 2022-12-12:
The console output for the first command:
[image]

while the second takes much time and reported the same message (Requesting List of world names) but finally worked.
[image]