osrf/mbzirc #231: CMake Error at CMakeLists.txt:121 (find_package):

criseyanginvaild, 2022-12-20:
when i was use " IGNITION_VERSION=fortress colcon build --merge-install " to build my space i get this error:

[image]

bye the way, when the fortress installed and i get this error:

[image]

criseyanginvaild, 2022-12-20:
sudo apt install ros-galactic-ament-cmake-pycodestyle

criseyanginvaild, 2022-12-20:
according this method to solve other same problems and will succeed

criseyanginvaild, 2022-12-20:
the second picture , i haven't got the answer . please , if you know ,tell me.   Thank you !

iche033, 2022-12-20:
the 2nd to last line in the second picture shows that you're using a local copy of `libQt5Core.so.5`? Can you installing and using the system installed version of Qt5?

criseyanginvaild, 2022-12-31:
okay, the question seems to be solved which the nvidia driver is bad and i use the ubuntu 20.04 commend driver

closed 2022-12-31 by criseyanginvaild